bigtreetech/BIGTREETECH-TouchScreenFirmware #770: [BUG] (short description)

DJDOUBLEG, 2020-06-08:
Missing this line in Configuration.h
#define ONBOARD_SD_SUPPORT
When i insert this line in config and reflash the fw. 
Not show onboard SD menu  in touch mode.
[image]

oldman4U, 2020-06-08:
Hello DJDOUBLEG.

Could you please let us know which board you are using and which version of the firmware (board and TFT).

With the latest version of the firmware, the board SD support has to be enabled in the board firmware and in the TFT firmware there is an auto detect when using Marlin:

_#### Onboard / Printer SD Card Support
# On Marlin Firmware, The TFT will auto detect Onboard SD Card
# Auto-detect is not available for other firmwares like Smoothieware
#Options: [enable: 1, disable: 0, auto-detect: 2]
onboard_sd_support:2_

Hope this helps, please let us know

DJDOUBLEG, 2020-06-08:
https://photos.app.goo.gl/am552RfGaLFBNcLS8

DJDOUBLEG, 2020-06-08:
Work it again. Reflash/recompie/edit config.ini...

oldman4U, 2020-06-08:
Sounds great. In case this issue has been solved for you, please close this ticket.

Thank you

closed 2020-06-08 by DJDOUBLEG