bigtreetech/BIGTREETECH-TouchScreenFirmware #771: [Q] (Your Question?) what is the max usb a capacity i can use

bramankersmit, 2020-06-08:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

Tygrys-1, 2020-06-09:
I have tested working 32GB, formated FAT32 working ok, and 64 and 128GB formated exFAT not working. The same applies to SD card reader. Maybe higher than 32GB, but still formated FAT32 will work, as the FAT32 can handle up to 2TB.

bramankersmit, 2020-06-09:
> I have tested working 32GB, formated FAT32 working ok, and 64 and 128GB formated exFAT not working. The same applies to SD card reader. Maybe higher than 32GB, but still formated FAT32 will work, as the FAT32 can handle up to 2TB.

closed 2020-06-09 by bramankersmit