bigtreetech/BIGTREETECH-TouchScreenFirmware #772: [BUG] Print stops almost at the end

VijitCoder, 2020-06-08:
### Description

LCD emulation mode.

I have a few sliced models that stops to print close to the end. The hotend/bed keeps heating, the  printer says that the printing in process but nothing happens. When I push the knob I got the normal menu which is means, that in fact the printing is ended.

### Steps to reproduce

Take the file from the [attachment](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4747891/files.zip). This gcode was prepared for Creality Ender 3.

Try to print it without filament (temp 195/0). It will take about 20 minutes.

**Expected behavior**

After all the hotend should be parking, hotend heating switching off, the bed moved to [0, 200].

**Actual behavior**

The printing just stops. No parking, no cold down.

### Hardware Variant

Bigtreetech TFT35-E3 V3.0

### TFT Firmware Version & Main Board Firmware details

Marlin 2.x
Screen BIGTREE_TFT35_V3.0.26.x

### Additional Information

I tried to change SD card. Doesn't help.

I tried to print this model with the stock LCD screen - works properly.

After all I plugged off the RS232 cable from the TFT screen and printed in emulation successfully. 

Something wrong in that place - data exchange by serial port. And this problem reproduces only with some gcode files.

[files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4747891/files.zip)

Tygrys-1, 2020-06-08:
Maybe the same as #735? I found actual version is working ok, maybe because of #730?

VijitCoder, 2020-06-08:
> Maybe the same as #735? 

I'm not sure that is the same. In my case no "out of service" from the printer. It works and reacts to the control knob. But not printing.

> I found actual version is working ok, maybe because of #730?

I use the version from the master. It not works. I checked just to be sure, I have the changes from the #730.

Did you try to print the file from my attachment?

VijitCoder, 2020-06-08:
I tried to print the problematic model with `#define SERIAL_ALWAYS_ON 0` and it works. So this is another sign that the problem is in RS232 exchange.

Tygrys-1, 2020-06-08:
Will try to print when my printer will finish it's current assigment. :-)

VijitCoder, 2020-06-09:
> Tomorrow I'll try to print the file from the Touch mode for more clarity.

Printed properly.

Tygrys-1, 2020-06-09:
Printed on Marlin 2.0.5.3, in Marlin mode, printer Ender 3 Pro/SKR mini E3 v1.2/TFT35 E3 v3.0,
On my printer it ended up smootly after 100% at 17:20 time. Head was parked to the left and bed moved forward. Printing with the air is funny ;-). The estimation progress jumped from ~70%+ to 100% on the end in a moment, but as the file has a large commented part after the gcode it seems ok.

VijitCoder, 2020-06-09:
Hm.. Well, I have no idea, why it happens with my hardware. In my case the bug reproduces constantly.

I'll later update the Marlin firmware and check this problem again.

VijitCoder, 2020-06-09:
@Tygrys-1 Could you provide your configs for TFT?

VijitCoder, 2020-06-09:
> the file has a large commented part after the gcode it seems ok.

That commented block prepared by Ideamaker. Another "pain in the neck", but not so critical :)

Tygrys-1, 2020-06-09:
Noticed this. Cura also adds some comments, but much less. 

Configs and FW:
[4.6.ZIP](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4752222/4.6.ZIP)

VijitCoder, 2020-06-09:
Thanks for configs. 

As I see there..
`serial_always_on:1` in config.ini
`#define SERIAL_ALWAYS_ON 0` in Configuration.h

I'm not sure which one is applied your TFT, but I know about another strange bug with this firmware: the device ignores configs from Configuration.h until you not reset all via Feature menu.

So, are you sure, that your print test was made in emulation with _a switched serial port on_?

Tygrys-1, 2020-06-09:
It goes like this: If no config.ini, the defaults from the configuration.h are used. You may also recall them through settings reset. If there is the config.ini file, it sets it's own set of settings. So the config.ini decides, how the things look like, but not all settings can be done through the config.ini, so the configuration.h still has it's role.

I can confirm, that my printer uses serial_always_on, as I have the PC connected through the WiFi to the TFT and without this setting the communication is somehow limmited in Marlin emulation.

VijitCoder, 2020-06-09:
Ok, I got it.Thanks.

Will return to this bug after Marlin update.

VijitCoder, 2020-06-09:
About configs. As I remember, Configuration.h ignored completely even if I change something via Feature menu. And the same by using config.ini. Any change by ini-file and Configuration.h will be ignored. And it doesn't matter even if via ini-file I change not the same as I specified in Configuration.h. I believe that is wrong. But I don't care about it while there is a lot of other problems.

Tygrys-1, 2020-06-09:
Configuration.h has less meaning after the introduction of the config.ini. The only case it is used is the "reset configuration" from the menu. (Or maybe after the change of the data format or corruption of the data storage). Through config.ini you may change settings just like in the menus. And some like the serial_always_on can be set only there, as there are no menu options for them.

roomme13, 2020-06-09:
i got this random issue too (and others issue #704) in Touch mode , just by navigating in menu sometimes TFT freeze, never back to normal, its hard to re-produce the issue.
even 5 min ago i turn on my printer, and it freeze from the start

Tygrys-1, 2020-06-09:
Hi roomme13,
It might be interesting, if in case of freeze you can go to the Marlin mode with the long knob press.
I also found, that when my TFT freezed, I still could access the printer through the WiFi. It might help diagnose the cause of the freeze.

gitxmax, 2020-06-14:
Same or very similar bug(s) with th skr pro board. Print constantly freezes with no errors. Tried multiple marlin versions, no joy ..
Guess BIGTREETECH's way to solve the problems is to release yet another board/screen, without bothering about the bugs with the current versions....

oldman4U, 2020-06-15:
Hi.

BTT could do a much better job - no question. Trying to clean the ticket system from unused tickets and therefore reading many many tickets i can see several issues which are not hardware related.

1. Users want everything - but have no experience (maybe advertisement makes them believe this is possible).
2. Much too much possible combinations - do you think that BTT should also support systems where the majority of the parts is not provided by BTT?
3. Bad documentation of available functions and how they work.
4. Willing to take -  but to give something back - NO. I see this as a human issue. Cheap hardware is great, free software - of course - and now i should do something for it!?

Most people complain about things but do not try to improve or even solve it. Seems to be also the case here in this community.

MY 2 cents.

Tygrys-1, 2020-06-15:
Kinda OT, but IMHO:
Reason 3 causes 1 and 2. And this should be made by BTT, not the community. At least maintained by BTT.
We* are only humans. And most of us just like to use things, not to maintain them. And most of us do not have enough resources or knowledge or mostly important - the time, to do a maintainance for a product made by a commercial company.

*) Actually, I'm the cat. :-)

VijitCoder, 2020-06-15:
> Willing to take - but to give something back - NO. I see this as a human issue. 

How about the money, is that not "give something back"? I donno where did you got your version of the hardware, may be the BTT just gifted it for you, but in my case I bought it on AliExpress. And there is no word that I must contribute to it after all. 

> Cheap hardware is great, free software - of course - and now i should do something for it!?

It not much cheaper then Lerdge hardware which have closed firmwares and supported by manufacturer, awaik. So _they guarantee_ that their devices will work. But in the case of BTT buyers (users) become beta testers and contributors. Something wrong here, I think.

VijitCoder, 2020-06-15:
> Most people complain about things but do not try to improve or even solve it. Seems to be also the case here in this community.

Most people of this community are 3d masters, but not the coders, I suppose. How they can improve or solve the hard/software problems of the printer itself? I see the only way - rollback to the stock hardware.

oldman4U, 2020-06-15:
> Kinda OT, but IMHO:
> Reason 3 causes 1 and 2. And this should be made by BTT, not the community. At least maintained by BTT.
> We* are only humans. And most of us just like to use things, not to [...]

Cat,

Agree to everything you wrote. About the order of the reasons we can discuss, no question.

I can see you responding to many tickets - so you are out of this discussion;-) but what you finally bought from BTT is the hardware. So if something does not work hardware wise, we have to blame BTT, otherwise Marlin and therefore ALL of us. But to understand this, someone has to know what Open Source means and it would be BTT's job to communicate this

@VijitCoder I haven't got anything from BTT or any other company. I bought everything by myself and if i collect all the things i bought in addition to what i really need, i can almost build a second printer;-) The time i bought my SKR Pro v1.1, a Google search for the board resulted in less than 40 hits. This makes you feel alone, believe me. But i bought it and started to build firmware using VSC (which i never heard before) and i learned to use this GitHub. In 2 months, I watched more YouTube videos than my whole life before and i spent sleepless nights, building hard and software. During this time I got a lot of help from other users, so I wanted to give something back, but i can not code and i can not engineer. So all i can do is to try to clean the system up and see if someone else can help with problems i can not help and to try to bring them together.

It is a community, so when you ask me what people like you and me can do (except to rollback to stock hardware) I have to ask you what YOU can DO and i am sure, we will find a way that you can add your skills to make this project a success.

But i can tell you one thing which will not improve this all. And this is to complain about problems without willing to help to solve them, thats for sure. But you know this and you do not act like this.

;-)

guruathwal, 2020-06-15:
You have not mentioned from which SD card are you printing the Onboard SDcard or TFT SDcard. 
Assuming it is TFT SDcard, It looks like a hardware issue as the most common cause for printer freezing/stoping while printing is due to communication error between the mainboard and the SDcard. SD card signals are very sensitive to interference. This may be due to loose SD card adapter, too long cable from SD card to mainboard, the cable is tangled/sandwiched between high current wires, or the cable has broken wires somewhere in between which may not be visible but some vibration might trigger it. 
The TFT SDcard is hardwired to the EXP ports. While printing through the emulator mode everything from reading the sd card to processing the gcodes is done by the mainboard. The TFT only displays the information just like a normal 12864 Graphic LCD.
You can try replacing the cables and separate the EXP cables from other wires to reduce interference.

VijitCoder, 2020-06-15:
From the TFT SD card. At that moment I even didn't know that the screen supports onboard cards.

VijitCoder, 2020-06-15:
Well, it might be a hardware problem, I'll not argue. But that problem reproduces every time _in the same manner_ from with a different cards. I think this not the case of interference or so. 

And as it was already mentioned above, when I switched off the `#define SERIAL_ALWAYS_ON` the bug dissapeared.

VijitCoder, 2020-06-15:
@oldman4U I got your point. All that you said relates to the community and I agree that this is a power. But my claim is not to community at all, but to the another player in the shadow - the Company. 

The company which sales the hardware and harvests the community work; makes the name and money to itself while the endpoint user buying so a raw device that it turns to pain for him.

And I convinced that is wrong. This is a cheating of the user. And the community has nothing to do with it. My accusation is only against the company.

oldman4U, 2020-06-15:
@VijitCoder I know;-)

For me the price of the products is the reason for most of the problems. More income, more money for development, support and maintenance. But there are companies doing exactly this, so every customer can choose the manufacturer and therefore the time and money he wants to spend for a certain solution. Going for the cheapest and expecting the best usually does not work and BTT is no exception.

guruathwal, 2020-06-15:
@VijitCoder my friend no one here is getting paid from BTT for anything. The money paid to buy the products is only for the hardware, not the opensource software. If the opensource contributions were paid by the companies then the donation concept would have not been invented. 
This repository receives most of the updates from the users. I myself have been providing my contributions and have not received any monetary benefits for the same. Majority of the issues here are resolved by the users like you and me. The ones who get paid by any company do not bother about anyone's problem. They are doing what they paid for. The ones trying to help are the ones who know how it feels when you can't get any help to solve your problems.
Just go through all the open and closed issues in this repository, you will see yourself who is helping who.

VijitCoder, 2020-06-15:
@guruathwal you don't understand me. 

I know, what is open source and of course the Company will not pay to contributors.

First of all: the company sales the hardware and earns the money on it. But without the open source software that hardware costs nothing. The company's success is based on the hard work of contributors for free. And I can live with that, but here is another, the worst moment: the company doesn't provides any guarantee to her buyers that the saled device will work properly. It's completely on the community.

Resume: the company makes money without any responsibility against the user. All negative is splashed out to the community, the ordinary user suffers but not the Company.

AnHardt, 2020-06-15:
A tiny bit off topic:
I want to thank BTT for taking the risk to solve some egg - hen problems by producing some boards and providing an open initial software environment to program these.
Especially for the SKR PRO V1.1 . For a good while it was obvious Marlin could make use off the power boost by the FPU and high clock-rates from a F4. SKR PRO V1.1 is far from a perfect board - not using the ideal pins for some functions of the board. The framework also is still far from perfect. But without this board the open source developers would not have had a ready, affordable, available , relatively open board to develop the F4-framework on. When BTT launched the board they did know there would be some problems - like with the USB-Host-Mode to read USB-sticks, and they did and do warn about that. But i don't think they really had a clue about how many unsolved problems there have been and still are. Later convincing a marketing department to include additional restrictions on ads is quite impossible for the ones knowing about the problems.

guruathwal, 2020-06-15:
Agreeing with @AnHardt BTT has made many great products. They are also good at resolving a customer's problem in comparison to other companies.
@VijitCoder I understand your situation. For any hardware related problem or any issues related to after-sales service, you should contact the company directly via email or on the company's Facebook page/group. They will respond to your query.
Github is not a place for this. I do not know why people post comments about any company on GitHub. GitHub is a development platform, not customer service.

oldman4U, 2020-06-15:
What if we focus again on improving what we like most, our 3D printers (hope my wife will never read this;-))

We all can make it better if we work together and the same time BTT can also focus on their part - developing new great hardware (and development costs are the most expansive parts of the hardware we buy) which we can buy for really small money. Those looking for a better solution are willing to pay more for it. For me, even i could easily afford it, this open source way is the better way and i say this after some frustrating situations.

VijitCoder, 2020-06-16:
I updated Marlin up to **bugfix-2.0.x 2020-06-14**. The problem is not reproduced anymore.

closed 2020-06-16 by VijitCoder