bigtreetech/BIGTREETECH-TouchScreenFirmware #773: [FR] Support Marlin dialogs in touch mode, like the pause or filament change.

Tygrys-1, 2020-06-08:
**Is your feature request related to a problem? Please describe.**
There is a need to interact with Marlin "dialogs" like the filament change or advanced pause.

TFT in touch mode acts like a remote serial terminal to the printer. That way it cannot do some interactive stuff like "continue" after the pause, or filament load/unload invoked from the gcode. Even if we can intercept this from the TFT's SD or USB file print, there is no way to "see" such situation on printer's SD print. It can also come from any serial port or WiFi module.
Might solve #760, #758 and others

**Describe the solution you'd like**
Maybe we can "catch" such situation, as the Marlin's display is connected and updating it's data even when the TFT is in the touch mode. 

**Describe alternatives you've considered**
It might require some changes in the Marlin, maybe it can send some special message on terminal or something like that and accept the "continue" gcode or emulated knob move/press.

**Additional context**
Lack of this feature limits the use of the touch mode as the printer daily driver and requires to do some things from Marlin mode.

dp250f, 2020-06-17:
TFT host mode needs to be able to use Marlin's HOST_ACTION_COMMANDS in order to have this functionality. Issue #736

ETE-Design, 2020-08-06:
https://www.youtube.com/watch?v=HJpfO8XifqI

bigtreetech, 2020-09-01:
Hello HOST_ACTION_COMMANDS should be worked after #984

closed 2020-09-01 by bigtreetech