bigtreetech/BIGTREETECH-TouchScreenFirmware #775: [Q] Why in the touch mode the screen doesn't see onboard SD

VijitCoder, 2020-06-09:
**Your question or details you need. Please describe.**

I suppose the screen should support a card inserted to the main board. But when in the Touch mode I go to the Print menu - there are SD and USB flash only. They are related to the screen ports. 

Why in the Touch mode the screen doesn't see the onboard SD?

In the LCD emulation the onboard SD works properly.

**specify the device variant related to your question.**

Bigtreetech TFT35-E3 V3.0 26.x
Marlin 2.x on BTT SKR-E3 Dip

Tygrys-1, 2020-06-09:
This is the "feature" ;-)
Just set the coresponding parameter in config.ini to "enabled".
onboard_sd_support:1 

I think, that the answer to your question is: Because the autodetect feature for SD was added, and this requires the Marlin bugfix version (whatever it was, definitly newer then yours). #627
Also #770

VijitCoder, 2020-06-09:
Yeah, my Marlin definitely older then 2020-04-27 :)

Thanks, I'll check it again after update.

Tygrys-1, 2020-06-09:
I also use older, but with this config setting I told the TFT, that I want my SD always visible and f* off the autodetect ;-)

VijitCoder, 2020-06-16:
I updated Marlin up to **bugfix-2.0.x 2020-06-14**. Auto detection works properly now. And even appeared support of the long files names, which is definitely nice.

closed 2020-06-16 by VijitCoder

ssombra, 2020-09-11:
I came here because I have the same problem with the SD card. However, I have some questions regarding the config.ini file: how does it work? should I set all parameters in the file and put it in the SD card root? or just those I want to change? Does the SD card needs to be there every time I turn the printer so the screen is updated with the config.ini settings? or the changes are permanent until I flash the screen again?