bigtreetech/BIGTREETECH-TouchScreenFirmware #776: Can't get the encoder to work after flashing firmware

hillsh, 2020-06-10:
This is  a new install of TFT24 with a RAMBO board.  I flashed the TFT24 with the new version of the firmware (1.26) and it all seemed to go well.  Before, in factory test mode, I was able to use the encoder to switch between touch mode and 12864 LCD simulator, but now the encoder doesn't do anything so I am stuck in touch mode unable to enter Marllin mode. The dial does work to go the the mode switch page, but turning the dial apparently does nothing.  Is there something I need to do in the Marlin configuration.h?

Tygrys-1, 2020-06-10:
Use the touch screen to touch the Marlin or BTT icon to choose the mode. Knob is not working on the mode selection screen anymore. I have no idea why.

hillsh, 2020-06-10:
Thanks for the reply.  So I did that, but the Marlin software is not being displayed - all I see is the line that says" 12864 LCD Simulator". I'm pretty sure that firmware loaded properly, but when I try to reload it nothing happens, either on power up or with a reset.  Any suggestions?

Tygrys-1, 2020-06-10:
Just turn or press the knob. Or wait some time.

hillsh, 2020-06-11:
I now have the marlin screen up.  I was enabling the wrong controller, I needed to uncomment *define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.  However, turning the knob does nothing - is there a specific setting for ENCODER_PULSES_PER_STEP and ENCODER_PULSES_PER_MENU_ITEM?

radek8, 2020-06-11:
If you have the EXP2 cable properly connected, the encoder should work.

hillsh, 2020-06-11:
I'm going to have to assume that the TFT24 is damaged, as the turning the encoder doesn't cause any changes to the menu.

radek8, 2020-06-11:
If the encoder works in touch mode and not in Marlin.
Measure the EXP2 cable for damage

radek8, 2020-06-11:
> Teď mám marlinovou obrazovku. Aktivoval jsem nesprávný řadič, potřeboval jsem odkomentovat * definovat REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. Otáčením knoflíku se však nic neděje - existuje konkrétní nastavení pro ENCODER_PULSES_PER_STEP a ENCODER_PULSES_PER_MENU_ITEM?

The encoder setting is as sensitive as you want it to be. I have it set
#define ENCODER_PULSES_PER_STEP 3
#define ENCODER_STEPS_PER_MENU_ITEM 1
and works well

hillsh, 2020-06-11:
Yeah, it's damaged I think.  I'm closing the issue.

closed 2020-06-11 by hillsh