bigtreetech/BIGTREETECH-TouchScreenFirmware #778: Updating MKS TFT24 v1.1 to firmware 1.1.26 does not work, the lcd just boots up with old firmware

klayus, 2020-06-10:
So i bought an MKS TFT24 and i'm trying to update to the latest firmware. The problem is i cannot make it to start the upload process except if i change the name of the bin file from BIGTREE_TFT24_V1.1.25.bin to mkstft24.bin but then the update process does not end. it hangs at 100% and just stays there. right now i can upload firmware tft24_V3.0.5 Release file with that folder structure but i'm pretty sure i'm missing something and can't figure out what. please help

oldman4U, 2020-06-12:
Hello klayus.

Seems your question has been answered, so could you please close this ticket.

Thank you

oldman4U, 2020-06-15:
Hello klayus.

It would help us and also other users like you, if unneeded tickets would be closed by the owner.

@bigtreetech Please close this unneeded ticket. Thank you

klayus, 2020-06-15:
hey sorry for not answering. seems i was wrong :) thank you for your help.

closed 2020-06-15 by klayus

karabas2011, 2020-06-17:
> Only the MKS TFT32 1.4 is supported in this firmware.

No. My MKS TFT32 v4.0 works. Only rotate of screen does not.

thisiskeithb, 2020-06-17:
> No. My MKS TFT32 v4.0 works.

Using firmware from this repo?