bigtreetech/BIGTREETECH-TouchScreenFirmware #779: Bug report 01 wrong command assigned to Init sd card button

B-rad888, 2020-06-11:
name: Bug report 01
about: wrong command assigned to Init sd card button 
title: 'Init sd card button does wrong command'
labels: 'bug'
assignees: ''BIGTREETECH”

Pressing the Init sd card button in the custom menu sends the wrong command. 
G28\nG29\nG0\n which can be found in the included config file on line 323 and 324

### Steps to reproduce

1. (In touch screen mode)locate Init sd card button either:
	Menu > custom > Init sd card
	Menu > settings > Machine > custom > Init sd card
(Why have two “custom” menus? It seems unnecessary, but ok. Either way both produce the same bug)
2. Press Init sd card
3. Watch in confusion 

**Expected behavior**

Pressing the Init sd card button > initializes the sd card

**Actual behavior**
Pressing the Init sd card button > Auto homes the printer > Attempts to auto bed level > produces an error because auto bed level is disabled. 

### Hardware Variant
TFT35E3_V3.0

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x 
SKR 1.4 Turbo
Marlin bugfix-2.0.x

### Additional Information
In Config file
custom_label_2:Init sd card
custom_gcode_2:G28\nG29\nG0\n

oldman4U, 2020-06-11:
Hi b-rad888.

Could you please compile the Master branch and upload it to your screen to see if the 4 reported bugs still exist then. 

Thank you

oldman4U, 2020-06-11:
@guruathwal 

Could you please check, thank you

[image]

oldman4U, 2020-06-11:
B-rad888

When you compile the Master Branch, you can fix this issue in the config.ini by yourself.

Thank you

B-rad888, 2020-06-11:
Thanks. I already did that before I posted this. I just wanted it to be known and fixed so others wouldn't have the same issue.

B-rad888, 2020-06-11:
downloaded the master branch like 3 days ago and flashed new firmware. Did something change?

PR #784 (2020-06-12): Fix wrong Gcode in Config.ini, 'T0' temperature updating to current tool/nozzle, Add parsing of Fan speed
- Add parsing of Fan speed
- Fix: 'T0' temperature updating to current tool/nozzle
- Fix: typo (raido > radio)
- Fix: wrong gcode in config.ini

Fixes #779
closed 2020-06-12 by bigtreetech