bigtreetech/BIGTREETECH-TouchScreenFirmware #780: name: Bug report 02 “Connecting …M115” Forever

B-rad888, 2020-06-11:
name: Bug report 02
about: Switching to marlin mode after power on TFT = “Connecting …M115” Forever
title: switching screens is not seamless
labels: 'bug'
assignees: '''BIGTREETECH”

Powering on the TFT screen works great in touch screen mode. Then switch to Marlin mode and you will see “Connecting…M115” while it never connects. Requires a reboot to connect. However, booting into Marlin mode first, then switching to touch mode and you won’t get the bug. This is not ideal as most users including myself would like to boot into touch screen mode, and have the ability to switch to marlin mode to make fine adjustments and go back to touch screen mode without having to reboot the printer. Until it is fixed seamless operation requires booting into marlin mode on startup and switching to touch mode. Using this work around will not produce the bug.

### Steps to reproduce

1. Power on TFT in touch screen mode
2. Wait for TFT to say ready
3. Switch to marlin mode either:
	Hold scroll wheel for 3 seconds and select marlin mode
	Press on screen for 3 seconds and select marlin mode
4. Turn scroll wheel to initialize screen in marlin mode
5.  Select Info screen
6. Stare at the connecting…M115 in confusion for eternity. 

**Expected behavior**
After powering on into touch mode and switching to marlin mode:

I would expect the screen to say “CoM115” and to work properly. If “serial_always_on:1” is set in config
or
I would expect the screen to say “Printer name Ready” and to work properly. If “serial_always_on:0” is set in config
**Actual behavior**
After powering on into touch mode and switching to marlin mode:
 Screen displays “connecting…M115” for eternity. Requiring a reboot. Or abandoning marlin mode and navigating back to the touchscreen.

### Hardware Variant
TFT35_V3.0_E3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x 
SKR 1.4 Turbo
Marlin bugfix-2.0.x

### Additional Information
This may have to do with an error I see after updating the config file. Unknown Command “MM115”
If there is an extra M in the M115 code somewhere it might be causing issues. I haven’t seen anything like that in the code however.

oldman4U, 2020-06-11:
do you know the distribution date of the bugfix-2.0.x version you use? Should be 2020-04-27 or later

B-rad888, 2020-06-11:
I'm not sure where to find the date. I downloaded it like 5 days ago.

radek8, 2020-06-11:
Is this what you downloaded?
Even today I can download the year old version. :-)

B-rad888, 2020-06-11:
@radek8  Im not sure what you mean by "this"? I downloaded marlin from the main download page. Marlin "Nightly" Builds > Marlin 2.0 with bug fixes
Website:    https://marlinfw.org/meta/download/
Link:     https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip

radek8, 2020-06-11:
Yes, this version is current

B-rad888, 2020-06-11:
@radek8 Thanks for checking! 

I'm new to this board and screen. I made sure everything was up to date with the install and reported the problems immediately. Hopefully these issues can be fixed. I really like the setup. However, if the TFT has limitations I might have to go back to the original screen for usability. Especially if I run into more bugs after installing power loss recovery, filament sensor, and all the other stuff I have planned.

radek8, 2020-06-11:
So touch mode works and freezes in Marlin mode?
Can you take a picture of the error?

oldman4U, 2020-06-11:
Hi.

I believe you did everything right. But there has to be a difference because i can not see the 3 issues - but must say that i do not use onboard sd with my skr E3 DIP V1.0 board.

B-rad888, 2020-06-11:
[image]

B-rad888, 2020-06-11:
@radek8 The screen is not frozen. I can still navigate

radek8, 2020-06-11:
Thank you.
Please attach files
Configuration.h
Configuration_adv.h
I have a similar configuration, I will match my settings.

B-rad888, 2020-06-11:
@radek8 Thanks!
[Config H & ADV.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4764713/Config.H.ADV.zip)

radek8, 2020-06-11:
If you make a home.
Then you switch to marlin, what will be on the display?
The display shows the last command ...

B-rad888, 2020-06-11:
Same result, So I did the following:
1.Power on TFT in touch screen mode
2. Wait for TFT to say ready
3. Auto home
4. Switch to marlin mode
5.Turn scroll wheel to initialize screen in marlin mode
6.Select Info screen
7.Stare at the connecting…M115 in confusion for eternity.

radek8, 2020-06-11:
Uncomment
#define M115_GEOMETRY_REPORT
and recompile Marlin

B-rad888, 2020-06-11:
Thanks for the suggestion. However, I just re flashed firmware and same result.

B-rad888, 2020-06-11:
do I need to also uncomment the following?

 //#define CNC_COORDINATE_SYSTEMS

radek8, 2020-06-11:
uncomment and edit
#define SDCARD_CONNECTION ONBOARD

uncomment
#define LONG_FILENAME_HOST_SUPPORT

This is not related to this problem, but it is better to change it.

radek8, 2020-06-11:
> do I need to also uncomment the following?
> 
> //#define CNC_COORDINATE_SYSTEMS

you leave this as it is

oldman4U, 2020-06-11:
I tested with commented and uncommented and it does not show the M115 in both cases. So there has to be something else causing this.

We all count on Radek now;-)

B-rad888, 2020-06-11:
I appreciate the suggestions! :) However, changing to #define SDCARD_CONNECTION ONBOARD now only allows marlin mode to access the onboard SD where before it was accessing the TFT SD. It also makes my other bug with accessing the onboard sd from the tft worse. It will freeze on the loading screen 95% of the time trying to access the onboard sd.

radek8, 2020-06-11:
This looks like a communication problem on RS232

B-rad888, 2020-06-11:
Hmmm.. So would that mean something is defective in the board or the screen?

radek8, 2020-06-11:
Both variants are possible

B-rad888, 2020-06-11:
Interesting. So I'm not sure what to do with my set up then. Is no one experiencing the bugs that I am?  Is there any way to determine the root cause? Should I as for a warranty replacement?

radek8, 2020-06-11:
the other errors you describe point to this

radek8, 2020-06-11:
It should be tried which board is defective, but it means reconnecting a few connectors and changing the configuration.
look here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/715

radek8, 2020-06-11:
Would you be able to switch the RX and TX pins to another serial port on the display to see if the display has a faulty RS232 input?

B-rad888, 2020-06-11:
I could swap the pins. However, I'm a little confused on how to do so at the moment. I'm still looking at the other post, trying to wrap my head around what actually needs to be done.

B-rad888, 2020-06-11:
Yes I'm not sure. The post is a little beyond me right now. This is my first time building code and doing a rebuild of a 3d printer.

guruathwal, 2020-06-11:
In **Marlin Mode** the TFT acts as a dumb monitor connected to a CPU (Mainboard). All the text you see in **Marlin Mode** is, in reality, happening in Marlin firmware on the Mainboard.

Have you also connected **OctoPrint** or **ESP3D** to the Mainboard? The `Connecting...` text is not coming from the TFT (unless any part of the TFT firmware is changed to do so). This message is more commonly used by other hosts controllers like (OctoPrint and ESP3D) when they try to connect to a WiFi network.

B-rad888, 2020-06-11:
I haven't connected octoprint, and I do have a wifi module connected to the tft

B-rad888, 2020-06-11:
@guruathwal I think you are on to something here. Removing the wifi module and doing the same steps as above actually changes the "connecting…M115" to "media inserted

1, remove wifi module
2.Power on TFT in touch screen mode
3. Wait for TFT to say ready
4. Switch to marlin mode
5.Turn scroll wheel to initialize screen in marlin mode
6.Select Info screen
7.Stare at the new display saying "Media Inserted"

[image]

B-rad888, 2020-06-11:
Here I was tearing everything apart. However I'm still wondering about accessing the media, how that works, and a possible hardware issue that @radek8 mentioned

radek8, 2020-06-11:
In this case, there will be no problem with HW.
It does not have WIFI configured, which tried to connect constantly.

guruathwal, 2020-06-11:
> I haven't connected octoprint, and I do have a wifi module connected to the tft

Ok, you did not mention any wifi module connected to the TFT controller in your issue description. Always mention full details of the hardware and software versions and variants.

The `Connecting …M115`  is displayed because the commands from the TFT and Wifi module are getting mixed up. Maybe because Wifi module is not sending EOL `\n` in `M117` Gcode while it tries to connect to a network.
You can ignore the message or you can report the issue in the Wifi module's GitHub page.

radek8, 2020-06-11:
What does the Marlin screen write to you after disconnecting the WIFI module and resetting the printer?

B-rad888, 2020-06-11:
Thank you all so much for all your help! The TFT really doesn't like the wifi module. After some more testing selecting onboardSD from the TFT works without freezing. It does take about 30 seconds to load but it works.  

My only other question is: In marlin Mode can I only have one media source available?  Either TFT SD or Onboard SD? I ask because I can mount media but not all three sources like in the TFT. 

@radek8  It displays media inserted like the picture above :)

B-rad888, 2020-06-11:
Also I'm really sorry for not mentioning the Wifi module. I kind of forgot about it, since I haven't used it. I just set it up and forgot about it.

radek8, 2020-06-11:
Yes, you only have one SD card available in merlin mode. That's why I recommended you to enable 2 parameters above.
In marlin mode you will print from onboard SD.
Onboard SD, TFT SD and USB will be available in touch mode.

B-rad888, 2020-06-11:
Perfect! It looks like everything except the Wifi is working correctly. I'll have to trouble shoot that some more. I can close these three open issues and combine them into one wifi issue if the problem persists. Thank you again I really appreciate all your help! :)

radek8, 2020-06-11:
I also bothered with WIFI, but if you know where to set it, it's a small thing ...

radek8, 2020-06-11:
If you have the same Wifi module, I will advise you

B-rad888, 2020-06-11:
Yes please. I followed the tutorial from Chri's basment on Youtube, but I think he advised to set the wrong frequency and installation method. I think that might be the issue.

radek8, 2020-06-11:
What module do you have and where do you want to connect it?
to the Marlin board or to the display?

oldman4U, 2020-06-11:
So Radek did all the work and Gurmeet showed us the right way;-)

I can also help with the WIFI, because I am using it in exactly this display. Unfortunately I can not help with the firmware, because I uploaded it using the SKR Pro I also have. So I can only help with the more simple part - but I can;-)))

Oldman

B-rad888, 2020-06-11:
I would like to connect it to the display because my case doesn't have a cut out. The module is the ESP-01S WIFI Module ESP8266 

https://www.biqu.equipment/products/esp-01s-wifi-module-esp8266-for-skr-pro?_pos=2&_sid=31dd870de&_ss=r

B-rad888, 2020-06-11:
@oldman4U  lol. Yes @guruathwal  came out of nowhere and solved it.

radek8, 2020-06-11:
It works even without a cutout.
Each variant has its + and -
[image]
[image]

B-rad888, 2020-06-11:
Interesting. I do have a spare ribbon cable from my old screen. I didnt think of that. I do have to reprint my case though so I could make a cut out. Its really whatever is most reliable I guess.

radek8, 2020-06-11:
For starters, put it on the screen, it will be easier

B-rad888, 2020-06-11:
That is true. I think it needs to be re-flashed though. I think the wrong setup is what cause all the issues.

radek8, 2020-06-11:
If you wanted to put it in SKR 1.4 turbo, you would have to change the port pins for serial port3 in Visual Studio.
Also, you could not use the USB Connector

B-rad888, 2020-06-11:
So, no octoprint then if its on the SKR 1.4 board?

radek8, 2020-06-11:
Marlin defines only 2 serial inputs. on one you have a display and on the other a USB connector or WIFI

B-rad888, 2020-06-11:
That makes sense

radek8, 2020-06-11:
The display expands another 3 serial inputs, which it sends to one

B-rad888, 2020-06-11:
So I already had this set up with the tutorial. The comment I saw on said, 

"I thought the ESP3D guys recommended running it at the 160mhz" 

and the response was... 

"i think they do, but there are so many different ones of these, I was going with the doc I found, honestly I have tried a bunch of different ways and can't see where it makes a whole lot of difference."

Would the wrong frequency be causing this? Because its all set up. I can control the printer and everything. However the TFT has all those issues when its connected.

Im logged into the web interface now and I cant see any setting to change that. Would I have to re flash the Wifi card?

B-rad888, 2020-06-11:
This is the tutorial: https://www.youtube.com/watch?v=NGgzw-XayEo

It was all set up through VS code and not the program to flash it

radek8, 2020-06-11:
napřed nastav Marlin dle obrázku.
SSID a pasword  nastav dle své wifi

[image]

[22 more replies]

closed 2020-06-15 by B-rad888