bigtreetech/BIGTREETECH-TouchScreenFirmware #782: Bug report 04 Marlin mode can only print from TFT SD

B-rad888, 2020-06-11:
name: Bug report 04
about: Marlin mode can only print from TFT SD 
title: Marlin mode can only print from TFT SD 
labels: 'bug'
assignees: ''BIGTREETECH”

Printing from marlin mode only pulls media from the TFT SD. Selecting change media does nothing. 

### Steps to reproduce

1.Power the screen into marlin mode. 
2. select print from media. ( only the TF SD will be avalible)
3. go back to main
4. select change media ( nothing happens) 

**Expected behavior**
I would expect the ability to change the media source. 

**Actual behavior**
Pressing change media does nothing

### Hardware Variant
TFT35E3_V3.0

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x 
SKR 1.4 Turbo
Marlin bugfix-2.0.x

### Additional Information
This really limits the ability of users to use everything correctly. If marlin mode can only use the TFT SD why would anyone want to use the USB option in touch mode? For consistency users would be more likely to use only the TFT SD. This may be a bug or just a limit of features. However, I hope this gets fixed.

Tygrys-1, 2020-06-12:
Cannot confirm that. From the Marlin mode I can print from the mainboard SD, the TFT SD is not seen in Marlin.

oldman4U, 2020-06-12:
Hi Tiger.

I believe all 3 tickets from B-rad888 have been related to a suspicious WLAN module.

@B-rad888 Could you please close all unneeded tickets.

Thank you

closed 2020-06-15 by B-rad888