bigtreetech/BIGTREETECH-TouchScreenFirmware #788: [BUG] Print does not end and can't be canceled

Dimitar-88, 2020-06-11:
Description
I encounter some issue here.

-After finish print from Onboard SD, i cannot go back to main menu, if i press back button. it show loading, but never reach to main menu until i reset the LCD.

-Also it stuck in loading if i attempt to stop a running print.

Hardware Variant
TFT35 V3.0
SKR Mini E3 V1.2

TFT Firmware Version & Main Board Firmware details
Latest build from master
https://github.com/roomme13/BIGTREETECH-TouchScreenFirmware/tree/build-xx.26.1

Additional Information
I use latest Marlin Bugfix2.0.x
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2

Both Config Files are attached
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4767219/Configuration.zip)

Tygrys-1, 2020-06-11:
By chancr - do you have the neopixel LEDs on the printer?

Dimitar-88, 2020-06-12:
@Tygrys-1 No i do not have neopixel LEDs. Just the motherboard, TFT35 and shutdown relay 1.3.

Tygrys-1, 2020-06-12:
Ok. In my case after the print, the printer requires one knob press to switch off neopixels. In this "end of print" state, the terminal is not working, and for that reason TFT waits with the "Reading SD" screen. After the knob press it resumes normal operation.

Ramoss007, 2020-06-12:
I have the same issue with V2.0 main board.
Cannot abort a print it lockes into Loading.... (seems to lost RS232 comms with main board)
When printing, Temp does not update, so no idea until print starts...
Going back to Marlin emu mode for now.

KL-XL, 2020-06-18:
I have exactly the same issues as the OP with mainboard BTT SKR Mini E3 V2.0 and BTT TFT35 E3 V3.0.

97turbotalon, 2020-06-20:
I'm also having the same issue with unable to cancel the print and locks on Loading screen.
BTT SKR Mini E3 V2.0 and BTT TFT35 E3 V3.0 using Marlin Bugfix-2.0.x

Dimitar-88, 2020-07-09:
It seems the issues got less after the last update. Now the auto shutdown works and machine turn's off at least . Temps get updated of course not while initial heat up but that's fine. Now layer or Height will be nice if it starts working and fan speed as well...

idjmic, 2020-07-30:
I'm running all the latest firmware (Marlin 2.0.6 and TFT 0.26.x) from July 30, 2020 on SKR mini e3 v1.2 with TFT35 e3 v3.0 only issue I've encountered so far is this one. I started a print from the SD Card (on motherboard) and tried to cancel right away... and it's waiting until it heats the bed before cancelling. Screen just say's Loading... and it does cancel the print after it finishes heating everything and goes to print... but the after she is FROZEN (reset here I come).

oldman4U, 2020-08-05:
@idjmic 

This issue is solved now. Please try

Ramoss007, 2020-08-05:
> @idjmic
> 
> This issue is solved now. Please try

Can you point me to the working firmware update please?

oldman4U, 2020-08-05:
Not sure how this can be done on the mobile. But it is the master branch in this repository

blueeagle69, 2020-08-16:
I just built from the latest Master source and the bug is still present.
Any ideas?

Thanks.

blueeagle69, 2020-08-17:
Hi.
Is there any updates regarding this?

Thanks 😀

oldman4U, 2020-08-18:
@blueeagle69 

Which one is still present. We had two issues initially.

blueeagle69, 2020-08-19:
> @blueeagle69
> 
> Which one is still present. We had two issues initially.

I'm printing from the internal onboard SD card.
When the print ends the display reads 98% complete and appears to stay there.
If I hit Stop I'm greeted with a Loading screen. 
And unless I'm mistaken I can get past the loading screen by clicking my encoder which is weird.
Printing from any TFT onboard source ends correctly.

(Hope this helps) 😀

oldman4U, 2020-08-19:
Ok. With this issue i can not help. Sorry

oldman4U, 2020-08-20:
But there is a pull request 984 which seems to be related to the issue with the print not finishing properly. 

Maybe you want to check

blueeagle69, 2020-08-20:
> But there is a pull request 984 which seems to be related to the issue with the print not finishing properly.
> 
> Maybe you want to check

Cheers I will take a look 😉

Thanks again.

oldman4U, 2020-08-26:
Seems there was another update today. 

Any luck with the new version?

blueeagle69, 2020-08-26:
Sorry for the delay.
I updated the TFT FW two days ago, so will test as soon as possible.

blueeagle69, 2020-08-27:
> Seems there was another update today.
> 
> Any luck with the new version?

Hi.

No the issue is still there when printing from the onboard SD.
Annoying.
And now the on-screen layer info has stopped working too.

oldman4U, 2020-08-28:
On screen layer stopped working after doing what?

blueeagle69, 2020-08-28:
Good question

> […](#)

Good question, it was working fine with my standard Prusaslicer Before layer change Gcode.
I flashed a FW update a week ago, and noticed there was no output on the Layer Info during printing.
It was working Okay before that, so answers on a postcard. 
I know nothing has changed in my slicer settings, so I'm stumped.

oldman4U, 2020-09-03:
I believe there is a (maybe closed) ticket related to you slicer and your problem. 

Marlin released 2.0.6.1 and of course there is also a new TFT firmware available. Maybe you could also try it without the shutdown item - just for a test.

blueeagle69, 2020-09-07:
Right, with the latest build the screen prints up a "Print Complete" popup and now appears to end correctly.
But the on-screen layer info is staying at 0.00 and not updating.

Thanks.

oldman4U, 2020-09-08:
@Dimitar-88 Please verify and close this ticket 

@blueeagle69 Please check all open and closed tickets for this new topic and start a new ticket if you can not find an answer. 

Thank you

Dimitar-88, 2020-09-08:
@oldman4U yep i will close it. For me it also works except layer height. My printer ia auto shutting off so all is good :)

closed 2020-09-08 by Dimitar-88

oldman4U, 2020-09-08:
Sounds great.

Maybe you can use your experience and help another user in the future.
Would be great. 👍🏻