bigtreetech/BIGTREETECH-TouchScreenFirmware #789: [Q] Why does my background have blue parts with a custom theme?

zech, 2020-06-12:
Hello,

I'm currently developing a custom theme for my TFT35 v3 and I'm facing some issues with blue color in parts of the screen where only `DarkGray` shoul appear.

[image]

I've appended my current Configuration.h here:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4769027/Configuration.zip)

Maybe someone can point me in the right direction on what I'm doing wrong.

Thanks in advance.
Tim

Wesie, 2020-06-25:
The background ICON must be matched to your backgroundcolor 
(like >>> ICON_BACKGROUND,           LABEL_BACKGROUND)
Also check, whichkind of color code was selected or chaged on your side..

zech, 2020-06-25:
@Wesie Thanks for your reply. Would you mind telling me where these options are?

I've configured "BACKGROUND_COLOR" as 15 (DarkGray) and my images have the same Color. As you can see from my photo only areas where no icon is shown are blue.

As attached in mein Configuration.zip these are my colors:

```c
/**
 * Default Touch Mode Color Options
 * Options: 0: WHITE,      1: BLACK,       2: RED,     3: GREEN,     4: BLUE,   5: CYAN,
 *          6: MAGENTA,    7: YELLOW,      8: ORANGE,  9: PURPLE,   10: LIME,  11: BROWN,
 *         12: DARKBLUE,  13: DARKGREEN,  14: GRAY,   15: DARKGRAY
 */
#define TITLE_BACKGROUND_COLOR     15  // Title background color // 0xD928
#define BACKGROUND_COLOR           15  // Background color // 0x0A29
#define FONT_COLOR                 0  // Font foreground color
#define REMINDER_FONT_COLOR        7  // Reminder font color, such as: "No print attached", "Busy processing", etc.
#define VOLUME_REMINDER_FONT_COLOR 5  // Volume reminder font color, such as: "Card inserted", "Card removed"
#define STATUS_XYZ_BG_COLOR        15 // Backgroud color for X Y Z positon display in Status Screen.
#define LISTVIEW_BORDER_COLOR      15 // Border color in List view
#define LISTVIEW_ICON_COLOR        15 // icon color in List view
```

15 means DarkGrey which is a constant for RGB565 `0x2124` which is in turn RGB888 #212421 an all my images have this as the background color.

[image]

Wesie, 2020-06-25:
YOu have changed the right area? There is in most of the case also a part in the setting for the pur text menue ;)

zech, 2020-06-25:
I did not change anything in my display settings. I've changed it only in code so far.

zech, 2020-06-25:
Is there something like an EEPROM saving settings from previous versions?

Wesie, 2020-06-25:
Independ which parameter you've changed. A couple of parameters (default) will be overloaded /changed from the setting which are make the user in the Config. I I.
You can modify this file and copy direct into the root directory. After reboot the new Config will be integrated and the settings are valid. But I couldn't remember for color settings in the Config. ini. 
Which firmware version does you use? 
Round about 7days before, there was a patch for better color integrated.... I've only look to the changes but not included in my modified sources...

zech, 2020-06-26:
I'm not using a `config.ini`

I'm compiling from source on `master` branch. So latest firmware I'd say.

zech, 2020-06-26:
Update:

I've narrowed it down a bit. My setup is fine until a popup message appears with a beep like:

`Fade Height is off`

After this the blue areas start to appear.

zech, 2020-06-26:
This video should show more details:

https://photos.app.goo.gl/ZtsB4sFM9ei6r7vW7

Wesie, 2020-06-26:
The reminder window (pop-up) changed the color ;)
Change also the colors for this window or set your colors back to your values and all things will be fine ;)

zech, 2020-06-26:
> The reminder window (pop-up) changed the color ;)

That's what I noticed too. But that is not the reminder. Reminder is the "No printer attached" on top which is yellow in my configuration.

> Change also the colors for this window

How? I cannot find anythin in `Configuration.h` to change this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h

> or set your colors back to your values and all things will be fine ;)

Same as above" How do I set/reset these? Your instructions are not clear to me. As I mentioned above I do not use anything other than the firmware which I compiled from source. No `config.ini` no Settings within Screen Menu.

All I did was switching images as described here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/readme/English/How%20to%20custom%20icon%202020-04-11.pdf and changing the colors in `Configuration.h`

I did not expect to change anything other than `Configuration.h` and the Images to get to a new theme.

Wesie, 2020-06-26:
There is one info field on the status screen and the Pop up reminder. Look
to the popup window (which pops up in your video)

zech, 2020-06-26:
Yes. Thank you again for trying to help. But this Popup is not created by me. The touchscreen firmware creates this popup window. There is no configuration option either to colorize this popup.

This popup appears now and then when configurations in Marlin are not set. But other than that I do not have any control over this popup window.

There ist no color definition for this window. How can I prevent this window to change background color to blue?

Wesie, 2020-06-26:
I will look a little bit later if I see some helpfull things there.... I
let you know

Wesie, 2020-06-26:
I think, the popup-reminder is called out of parseACK.c
`   popupReminder((u8* )info, (u8 *)dmaL2Cache + ack_index);
`
You can try if you change it too:

>  popupReminder((u8* )info, (u8 *)dmaL2Cache + ack_index);
>  GUI_RestoreColorDefault();

I don't know if its work, because I have only see this call, don't use it direct during change the code.
The reason for different color I see here (Popup.c)

> WINDOW window = {
>   POPUP_RECT_WINDOW,                      //rectangle position and size of popup window
>   10,                                     //Four-corner arc radius
>   3,                                      //Outer line width
>   0x5D7B,                                 //Outer and title bar background color
>   {MAGENTA, 0x5D7B, POPUP_TITLE_HEIGHT},  //Title bar font color / background color / height
>   {WHITE, BLACK, POPUP_TEXT_HEIGHT},      //Message area font color / background color / height
>   {WHITE, GRAY,  POPUP_BOTTOM_HEIGHT},    //Bottom (font color) / background color / (height)
> };

zech, 2020-06-26:
Ok. Then it's a Bug and not a configuration issue. I'll try that later.

moerchel, 2020-06-29:
I have the same issue. The code checked out before 2020-06-03 works as expected. Maybe some changes made in PR #755 (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/755) cause this problem.

bigtreetech, 2020-07-28:
Hello, I've just tested it, but I haven't reproduced this situation. Can someone confirm if the latest firmware still has this problem?

closed 2020-08-19 by bigtreetech