bigtreetech/BIGTREETECH-TouchScreenFirmware #793: TFT35 KNOB

SKOREIKO, 2020-06-15:
Im Pretty new with this, but i see some people with there tft35 and the knob is lit up,
can someone explain how i do this ?

SeanLMH, 2020-06-15:
If I recall this correctly it's menu -> settings -> feature. There should be an option there to set the knob colour. If it's not there, you may need to update your firmware on the TFT35.

SKOREIKO, 2020-06-15:
thanks for the response, when i do that i get a unknown command "m150 r255 u0 p255"

SKOREIKO, 2020-06-15:
sorry it was m150 R0 U255 B0 P255

Tygrys-1, 2020-06-15:
Not that setting. You are trying to set the printer's neopixel, which your printer might not have.
Do exactly as @SeanLMH said and search for the "Rotary knob LED" setting. You have to press down icon to browse the feature menu.

SKOREIKO, 2020-06-15:
thanks @SeanLMH and @Tygrys-1 ill have a look.

SKOREIKO, 2020-06-15:
i found a older version that works, i guess the newest doesn't , but thanks again

oldman4U, 2020-06-15:
Hello SKOREIKO.

There is a bug in v26.x which is fixed in Master already and which will be part of the next regular release.

Please close a ticket in case you do not need it anymore.

Thank you

closed 2020-06-15 by SKOREIKO