bigtreetech/BIGTREETECH-TouchScreenFirmware #794: Filament run out sensor works with marlin mode not working with tft

B-rad888, 2020-06-15:
### Description
After setting up my new filament sensor (BTT SFS V1.0 Smart Filament Sensor) I was able to get it to work pretty good **in marlin mode** with the following changes:

Plugged sensor into SKR 1.4 Turbo Pin : P1_26  // E0DET

Config.h
#define FILAMENT_RUNOUT_SENSOR
#define NUM_RUNOUT_SENSORS   1  
#define FIL_RUNOUT_INVERTING false
#define FIL_RUNOUT_PULLUP 
#define FILAMENT_RUNOUT_DISTANCE_MM 7
#define FILAMENT_MOTION_SENSOR
#define NOZZLE_PARK_FEATURE

Then I tried to get the sensor to work in **touch mode** with the following in configuration.adv:

#define M114_DETAIL  

It didn't  work in Touch mode. There are 3 settings in touch mode which can be activated, each with their own problem

Filament sensor setting : Problem
1. off: Printing starts normally, when a snag is detected it stops, error box pops up stating printer is 
    paused, heaters eventually warm back up, cant resume print, error box continually pops up saying 
    printer is paused, no matter how many times ok is pressed, error box continues to pop up. 
2. on: error box pops up immediately behind icons. Cant really see error, but might say Ftt, cant 
    print
3. smart: Filament sensor will trigger immediately, print can be resumed just to trigger again.

### Steps to reproduce

Listed above

**Expected behavior**
Perfection - no failed prints 

**Actual behavior**
limited capability, failed prints if sensor is set to off and triggered

### Hardware Variant
SKR 1.4 Turbo
BTT TFT35-E3 V3.0
BTT SFS V1.0 Smart Filament Sensor

### TFT Firmware Version & Main Board Firmware details
Marlin bugfix-2.0.x (newest version)
BIGTREE_TFT35_V3.0_E3.26.x (newest version)

### Additional Information
* Provide pictures or links to videos that clearly demonstrate the issue.

[Config H and ADV.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4777562/Config.H.and.ADV.zip)

Tygrys-1, 2020-06-15:
It actually works, but the TFT in touch mode does not "see", what is displayed on the Marlin's LCD. Some messages are displayed only there and are not send through the serial connection for the terminal (and TFT in touch mode is the terminal to the Marlin). You might be able to blindly use the knob and let the print to continue.
#760, #773

Your setting should be "off", as you do not have the sensor connected to the TFT board, and theese settings are for that case.

While printing from the Marlin SD card, the printer's filament sensor will pause the print, but with printing from the USB or TFT's SD card, it will be ignored.

oldman4U, 2020-06-15:
Tygrys-1 = #1

;-)

B-rad888, 2020-06-15:
I was hoping that wasn't the case. Well how about this?
[image]

B-rad888, 2020-06-15:
Wait wait... Version 2 with fully upgraded corrected wiring.. I think. Maybe
[image]

Tygrys-1, 2020-06-15:
IMHO you should not wire GND and VCC to both boards, definitly not the VCC. You may create the ground loop or burn something. The sensor should be powered from the mainboard. Only the blue wire is important. If this will be the on/off sensor, you will be able to hardwire the blue to the both boards. I do not know the nature of the "inteligent" one, but if it only outputs data, you will be able to hardwire it the same way.

This, however will not solve the case, where the print is done from the printer mainbord's SD card, and the TFT will be in the touch mode, as you still will not see the information from the Marlin about filament change, and will have to blindly use the knob to continue. #758

B-rad888, 2020-06-15:
Thanks for the advice. I was 1/2 way joking. Although I think something like pictured above could be ridiculously practical. I'm trying to do a Bigtree tech build with all of the add-ons. 

1x SKR 1.4 Turbo
1x BTT TFT35-E3 V3.0 
1x Wifi (gave up on that one)
2x BTT SFS V1.0 Smart Filament Sensor 
1x BTT UPS 24V V1.0
1x BIGTREETECH Relay V1.2 Automatic Shutdown
1x BTT TF/SD Cloud (Possibly)
1x BL touch

I'm trying to get everything as compatible as possible and still use the touch screen. However every add on I make, makes the touch screen less compatible. I just installed the UPS and I cant even get that to work on marlin mode (not giving option for continue print). 
Link for bug I'm having with that one: https://github.com/MarlinFirmware/Marlin/issues/18307

Is it even worth it to get the filament sensor on the touch screen with everything else I want to add? 
I know there are 4 connections on the back. Possibly for all the add ons.

Its like a 2 day event just to under stand these things individually and get them working.

Tygrys-1, 2020-06-15:
Wifi is ok with the TFT35 E3. Why 2 sensors?
The Relay might have the same issue as the sensor - if you connect it to the mianboard, it will work in Marlin mode, or when printing from the mainboard SD. If you want it to function from the touch mode for USB/TFT SD print, you will have to connect it to the TFT. I have no idea about UPS.

The main missconception people have with BTT TFT touch mode is, that it will work like the printer's native display, as seen on other printers with touch screens. It will not. It will work like a terminal to the Marlin, with all consequences of this fact.

carl1961, 2020-06-15:
The filament sensor is just a optical sensor,  with a wheel with many
holes. Marlin has the code to know to want a set determined amount of time.
The op use 7 , so after 7mm of filament passes without the encoder changing
signals it triggers a filament out alarm. The tft does not use the Marlin
code, they been working on the encoder logic but btt has not got it right
yet, as far as I know. I just run in Marlin mode and I have my encoder set
at 25mm .

B-rad888, 2020-06-15:
I got the wifi to work kind of but it gave me other issues: see #780 .  
Also 2 sensor for dual extruders, dual colors, or materials. Just to be ridiculous i guess. Also it sounds really cool :)

I'm not to sure about the UPS either. It might be a marlin thing. I might need an older version. Or give up on that one too. its been giving false positives while sitting idle. On top of no option to resume. It does raise the gantry though. so no zits on my 1/2 finished prints. :) 
  
I just really like the interface of the touch screen, but I wouldn't mind just using marlin mode for more compatibility.

Tygrys-1, 2020-06-15:
Cool idea with 2 colors, ade you planning to use 1 mixing printhed or 2 separate nozzles?
I see no way to connect 2 inteligent sensors to TFT35. In case of simple ones, the diode logic can be used to have them connected.
I use WiFi on TFT35 with no problems, BTT SKR mini E3 v1.2 on the printer.

Consider adding the camera to the set :-)

B-rad888, 2020-06-15:
Thanks! I'm planning on a 2 in 1 out setup. Partly because the part cooling fan is on HE1 and partly for no oozing on the print. it would be interesting to try dual nozzles for dual temperatures for different materials. However, It does add complexity for alot of things and issues like the part cooling control. I know the SKR 1.4 is basically set up for 2 filament sensors based on the pin layout for E1det and E0det already being labeled run out sensor 1 and 2. Marlin also has options for more than one sensor so it might be best to stick with marlin mode.
 
lol I'am considering the camera as well (Raspberry Pi High Quality Camera). I have a raspberry pi 4 all set up with octoprint. I really wanted to enable power loss recovery (which Octopi doesn't support) and see if I could use Octolapse as a standalone uploading/monitoring/recording device. I'm not sure it would , but......That would be really nice! :)

Tygrys-1, 2020-06-15:
2 in 1 looks simple to implement, but every color change will cost you some filament. On the other hand you may mix colors.

Why not consider 3 extruders and RGB filament? ;-)))

Take a look at the Teaching Tech channel on YT (if you haven't already). The Octoprint/Octolapse and other octo-whatever plugins are explored there. 

I am using spare security camera to look at the print and the Marlin status screen at the same time and TFT wifi to eventually ask the printer for something. Octopi maybe in the future.

B-rad888, 2020-06-15:
lol 3 extruders I wish. Not enough drivers, unless i went with a 3dchameleon setup. RGB lights though are something I was going to put on the list eventually. 

ha ha yea I already maxed out my Octoprint add ons. Its really nice I like it. I was using the Pi as a kodi box until recently. I like it much better for octoprint. Its really snappy.

Tygrys-1, 2020-06-15:
[image]

B-rad888, 2020-06-15:
Cool! Looks good! :) How did you get the layer info and stats on the screen?

B-rad888, 2020-06-15:
Well its been fun! I guess we got off topic a bit and its way passed my bed time. I appreciate all the great info @Tygrys-1 . I'm going to close out the ticket for @oldman4U  since its not a bug and a feature. Thanks again everyone for the learning experience! :)

closed 2020-06-15 by B-rad888

Tygrys-1, 2020-06-15:
The layer info is from the Cura, in Post processing I enabled "Display Filename And Layer on LCD", and it adds M117 with this info between layers to the gcode. Do not try it with the touch mode, or try and forget about it ;-)
Other stats are from configuration_adv.h:
#if HAS_GRAPHICAL_LCD && EITHER(SDSUPPORT, LCD_SET_PROGRESS_MANUALLY)
  #define PRINT_PROGRESS_SHOW_DECIMALS // Show progress with decimal digits
  #define SHOW_REMAINING_TIME          // Display estimated time to completion
  #if ENABLED(SHOW_REMAINING_TIME)
    #define USE_M73_REMAINING_TIME     // Use remaining time from M73 command instead of estimation
    #define ROTATE_PROGRESS_DISPLAY    // Display (P)rogress, (E)lapsed, and (R)emaining time
  #endif
#endif

I think there is another setting to set to have percentage displayed, but cannot find at the moment.
If you will not get good results with the above, let me know and I'll dig further

oldman4U, 2020-06-15:
Good "boy";-) Nothing is off topic as long as it is 3D printer related!!!

Thank you

abecaw, 2020-12-17:
I have it working in touch screen mode as well as simulation mode and optioning. I have a Ender 3 pro with the BTT SKE mini E-3 V-1.2 mainboard. I have the TFT35 E3 V3.0. I have a Micro Swiss Direct Drive all metal hot end. Oh and a BlTouch. I'm using the latest version of Kay3D Cheetah and Marlin 2.0.7.2. I'm using PT-DET connector.

1CM69, 2021-04-14:
> I have it working in touch screen mode as well as simulation mode and optioning. I have a Ender 3 pro with the BTT SKE mini E-3 V-1.2 mainboard. I have the TFT35 E3 V3.0. I have a Micro Swiss Direct [...]

OK, how are you connecting the sensor to the SKR Mini & the TFT?

abecaw, 2021-04-17:
I’m sorry, I’ve been working on something else and not able to retrieve mail. Did you get this fixed?
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

1CM69, 2021-04-17:
> I’m sorry, I’ve been working on something else and not able to retrieve mail. Did you get this fixed? David M. Fox ABEC(AW) USN Ret.
> […](#)

Not yet, I’m in the process of setting this up and realise that it quite simple to have the sensor fitted to either the mainboard or the TFT but. It to both. 

If the sensor is fitted to the mainboard and I use LCD mode it works. It if I switch to TFT Touch mode the sensor does not work because it needs to be connected to the header on the TFT. 

How gave you got around this issue with the wiring connections & the coding in Marlin ??

Thanks in advance

abecaw, 2021-04-20:
Make sure your TFT has the latest firmware update.   
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

1CM69, 2021-04-21:
> Make sure your TFT has the latest firmware update. David M. Fox ABEC(AW) USN Ret.
> […](#)

Yep, fully updated firmware but you still have not explained how you have your sensor connected to both the mainboard and the tft at the same time.

The config file for the tft firmware specifically states that the PSU, UPS & Filament sensor setup is '(if connected to TFT controller)'
so although the options are available in the tft, they will not function if the sensors are only connected directly to the mainboard.

Do you have schematics for your wiring to both the mainboard and the tft board?

oldman4U, 2021-04-21:
I believe this is a misunderstanding and the sensor works does not matter if it is connected to the mainboard OR TFT. Just to make sure. There is no need to configure the sensor in Marlin in case it is connected to the TFT.

1CM69, 2021-04-21:
OK, so abecaw actually meant that the sensor/s work fine in TFT & Marlin if they are connected to either TFT -OR- Marlin, not in fact that they are connected to both simultaneously.

oldman4U, 2021-04-21:
I do not know but believe so. Yes.

abecaw, 2021-04-21:
Mine started working after I updated firmware.   
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

abecaw, 2021-04-21:
I have only connected it to mainboard NOT TFT.   
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

1CM69, 2021-04-21:
> I have only connected it to mainboard NOT TFT. David M. Fox ABEC(AW) USN Ret.
> […](#)

Thanks, misunderstanding sorted out now 👍🏻

ant0nyk1ng, 2021-04-21:
> 
> I have only connected it to mainboard NOT TFT. David M. Fox ABEC(AW) USN Ret.
> […](#)

abecaw what did you do in the TFT settings? What port on the motherboard did you use? After firmware update I can still only use it in Marlin Mode.

KingBain, 2021-04-26:
This ticket is very confusing.

So I have a SKR 1.4T and a TFT35.
Currently using BTT Touch Mode
Filemant sensor is connected to SKR mainboard.

Ifi im reading everything correctly.

My filament sensor will not work in touch mode because I have the sensor connected to SKR board.

Correct ?

1CM69, 2021-04-26:
> This ticket is very confusing.
> 
> So I have a SKR 1.4T and a TFT35.
> Currently using BTT Touch Mode
> Filemant sensor is connected to SKR mainboard.
> 
> Ifi im reading everything correctly.
> 
> My filament sensor will not work in touch mode because I have the sensor connected to SKR board.
> 
> Correct ?

Yes, you can have it connected to either board or tft but it’ll only work in the one device you have it connected to at that time.

KingBain, 2021-04-26:
if I switch out of BTT touch mode, will the filament sensor start to work ?

ant0nyk1ng, 2021-04-26:
I have it working in both Marlin mode and Touch mode with the latest release. 
Runout sensor plugged and config to X MAX. 
It works with SD card for both modes but I did not test USB or Onboard micro SD for Touch Mode.
In the touch mode I have the options turned off I think for runout sensor... I am not sure if that maters.

KingBain, 2021-04-27:
@ant0nyk1ng could you detail your config that's allowing that ?

arielito-ear, 2021-06-14:
Hello, I have the same problem, but my configuration is:

**- MTB TFT35 v3.0
- SKR 1.3
- Filamente sensor connected to SKR 1.3 board, PIN 1.29
- Firmware TFT35 v3.0: BIGTREE_TFT35_V3.0.27.x.bin
- Firmware SKR 1.3: Marlin Bugfix (Version 2.0.9, according to the configuration.h file)**

1. I have tried various combinations of the Marlin configuration, and the M119 command always behaves correctly in the absence or presence of filament.
2. Testing from Repetier Host, it also behaves correctly as the configuration changes with the M412 command.
3. I have tried all the possible combinations of the M412 command, along with the configuration of the TFT35 v3.0 screen, plus the different combinations of Marlin. And it has always worked fine from Repetier Host.
4. I have repeated the previous point but with the TFT35 v3.0 screen, and I have always obtained the following, regardless of the marlin configuration, and the M412 command:

- Filament sensor **ON**: With or without filament, it always says that filament is missing and cannot continue printing.
- Filament Sensor **OFF**: It never detects whether or not it has filament (_I guess this is correct_).
- Filament Sensor **SMART**: It starts printing and after a few seconds, it says that there is no filament. When pressing "Resume" it continues but says again that there is no filament (This regardless of whether or not there is a filament). And the same thing always repeats.

Just now I have found this Post, and I have tried with the screen in Marlin mode as mentioned by **_B-rad888_**, and it actually works correctly in that mode.

**What do I want?:**
Use the TFT35 v3.0 display, in Touch mode, with the sensor connected to the SKR 1.3 board. Nothing else.

Could anyone please help me with this? Or is it not possible to use the BTT TFT35 v3.0 display in this way?

Thanks!!!.

calderon67, 2021-06-28:
I have just the same issue with SKR 2 and tft35 v3

kisslorand, 2021-06-28:
Do you guys use M75 in your gCode as it is already written a hundred times in filament sensor connected to mainboard related topics?

calderon67, 2021-06-28:
Yes, I do

labotecno, 2022-04-09:
hello have some solution this ? i have the same issue

calderon67, 2022-04-09:
By the moment I have not found anyway to solve it. If you find the way to make it work, please, let us know

labotecno, 2022-04-10:
> By the moment I have not found anyway to solve it. If you find the way to make it work, please, let us know

i see the normal way ui lcd its working i get triggered run out  sensor with M119  but i  i see on the config file its important to put M75 for start  gcod and M77 for end gcod !
 `# NOTES for users having a filament sensor connected to the mainboard:
#   1) Enable the start/end G-code below.
#   2) Add the following commands to the start/end G-code:
#      - start_gcode: M75\n
#      - end_gcode: M77\n`

yecine06, 2022-08-10:
I tried adding the gcode to the start and end with my slicer to test it and it's not working.

Anyone figured it out ?

gac123, 2022-08-11:
adding M75 at the start of your gcode and M77 at the end works!

I have my creality filament sensor on the skr mini e3 board and I use touch mode, before it was indeed only able to work in marlin mode, which doesn't support usb files from the tft screen, thanks @labotecno !