bigtreetech/BIGTREETECH-TouchScreenFirmware #795: Not displaying Cuatom_Machine_Name

Mick-Green, 2020-06-15:
Hi

I've recently bought an SRK 1.4 (non turbo) board and it came with the TFT35 v2 screen. (updated to latest firmware on first use= Distribution date: 2020-05-12)

i have no problem setting it up and SKR 1.4 and TFT35 v2 are working ok together, but i cannot get the screen to display CUSTOM_MACHINE_NAME

i've looked in configuration.h and i can only see the following that would enable this so i uncommented it recompiled and uploaded still and the screen still just says Ready, not 3D Printer ready as i was expecting it to display

#define CONFIGURATION_H_VERSION 020006 "Also compiled using 020005 with same result"

// Name displayed in the LCD "Ready" message and Info menu
#define CUSTOM_MACHINE_NAME "3D Printer"

am i missing something obvious.

the string in Language,h

#ifdef CUSTOM_MACHINE_NAME
#undef MACHINE_NAME
#define MACHINE_NAME CUSTOM_MACHINE_NAME

refers back to the line in now uncommented in Configuration.h

Marlin 
Thanks michael.

Tygrys-1, 2020-06-15:
It is unclear to me, where do you expect this name to show up. You should see this in the Marlin status screen, and I think on Marlin status line after the print.
Do not expect any of theese in touch mode.

Mick-Green, 2020-06-16:
Hi 

I was expecting it to show up where the firmware says it would show up. in the information window on the display "printer name" READY.  instead of just saying ready

Is this not the case when you just have a touch screen. 

But if it is not supported in touch mode then that's OK. thanks for the response :-)

My printer dose not facilitate using an LCD screen with a Buttons for use in Marlin mode. I could however design something that i could mount a tft35 v3 screen in, but i think i will just forget about the screen and get a raspberry pi4 and go down the ocoprint route

bigtreetech, 2020-08-10:
Hello, This function should have been completed in #952,  please confirm.

closed 2020-08-19 by bigtreetech