bigtreetech/BIGTREETECH-TouchScreenFirmware #796: [FR] Manual Mesh Levelling TFT35 V2

stevegb666, 2020-06-15:
**Describe the solution you'd like**
I would like a page of buttons that allow me to complete manual mesh levelling through the tft without needing to connect to a pc.

I.e one that starts manual mesh levelling (G29 S1), next point (G29 S2) and move Z axis up or down by a small step.

I think it would be much more useful than the existing manual levelling page that just moves the axis to 4 set points.

If anyone has any idea how I can achieve this I'd be very grateful.

**Describe alternatives you've considered**
At the moment I have to connect up and use Pronterface

Tygrys-1, 2020-06-15:
If you install the BL Touch, there will be a nice support for G29 and automatic bed leveling, this menu is invisible if there is no probe installed. No mesh editing so far.
As the bed leveling requires interaction with the printer itself, it will be hard to implement it in touch mode, where the display is the serial terminal to the printer and lacks the interaction possibilities. The best way I see is to use Marlin mode for this.
You can invoke G29 (or any other gcode) from gcode list or built in terminal editor, so you do not need the PC to do this.

stevegb666, 2020-06-15:
Thanks,
I can run the g29 through the built in gcode but without an encoder it's hassle to actually move the z up and down by small incriments. Have to back out go into the move menu etc. Then go back to gcode again.

Without soldering and a lot of tweaking I didn't think it was possible to get 'true' marlin mode on the v2 just unified?

Tygrys-1, 2020-06-15:
I think there is the possibility to integrate the z move buttons with G29 dedicated buttons in some kind of menu to make things easier in that case. 

I was playing with manual leveling at the beginning of my adventure with printing, but since I bought probe for about 10$ I think it is not worth the hassle. Now I have the bed measured with about 2cm grid for imperfections, and at the beginning of each print I have 3 point check to tilt the mesh according to it's actual state. With manual mesh I was using 3x3 grid, and it was highly inacurate and time consuming to adjust the leveling with the sheet of paper, as I was never sure if the tension at actual point was the same as in previous. All frustration has gone after instaling and tuning the probe, so if your hardware supports it, I highly recomend this.

karabas2011, 2020-06-17:
> **Describe the solution you'd like**
> I would like a page of buttons that allow me to complete manual mesh levelling through the tft without needing to connect to a pc.
>
> I.e one that starts manual [...]

https://github.com/rebel1/BIGTREETECH-TouchScreenFirmware
It seems this fork does MBL

stevegb666, 2020-06-17:
Hi, I'd looked at that before but think it's only implemented on the v3 unless I'm missing a trick?
I compiled it for the v2 and didn't see any difference in the menus

stevegb666, 2020-06-18:
Much appreciated Karabas, went back and retraced my steps compiling it and that does indeed add exactly what I was looking for.

closed 2020-06-18 by stevegb666