bigtreetech/BIGTREETECH-TouchScreenFirmware #797: [Q] (Touch screen does not work after firmware load)

RavenDwn, 2020-06-15:
I bought the tft24 V1.1 and downloaded the latest firmware. After firmware loaded it went to touch screen configuration but would not give an accepted. While attempting to configure I hit the reset button and now touch screen does not work.
I tried to add a reset.txt file but that did not seem to help. 
Is there a way to force a touch screen test in the config file or unload the firmware and start over?

Tygrys-1, 2020-06-15:
When I downgraded firmware to previous version for testing (I think 25.something) it required the screen test again, and after upgrading to newest it also asked to calibrate the screen, so maybe this will do.

RavenDwn, 2020-06-16:
Thanks. Any idea where to get previous version

Tygrys-1, 2020-06-16:
From Github :-)
Here is [25.2](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/2e83dc384936ae8faa86a4be9bc5b33eaee98728)

karabas2011, 2020-06-17:
> I bought the tft24 V1.1 and downloaded the latest firmware. After firmware loaded it went to 
MKS TFT or BTT TFT?

RavenDwn, 2020-06-17:
I was able to use the version 25.2 firmware then reloaded update 26 and screen is working fine now THANKS ALL
Still can not get it to load printer. Updated the firmware for SKR V1.4 board and reloaded firmware for tft but still no luck. Still get "No Printer Attached" message

Tygrys-1, 2020-06-17:
Please check your connection speed. It should be the same on MB and TFT. You can set it in Settings on TFT.

radek8, 2020-06-17:
In Marlin you should have serial port 0 enabled
#define SERIAL_PORT 0
or
#define SERIAL_PORT_2 0

oldman4U, 2020-07-05:
Hi rav. 

Any news regarding the reported issue. Please close the ticket if you do not need it anymore

RavenDwn, 2020-07-05:
Thanks for all the help on this. Unfortunately it was a bad TFT board so I had to get a new one and decided to upgrade to TFT35. Got the new one yesterday and it hooked right up to MB and read printer as it should have. Thanks again

closed 2020-07-05 by RavenDwn