bigtreetech/BIGTREETECH-TouchScreenFirmware #798: [Q] Firmwares for BIGTREETECH TFT43 / TFT50 / TFT70 V3.0 ?

Guilouz, 2020-06-15:
Hi,

Where I can find new firmwares for BIGTREETECH TFT43 / TFT50 / TFT70 V3.0 ?

[image]

[image]

oldman4U, 2020-06-15:
Hi.

Do you have such a hardware?

Guilouz, 2020-06-15:
> Hi.
> 
> Do you have such a hardware?

I have order it.

max5962, 2020-06-16:
is preview model will be available on other screen ?

Guilouz, 2020-06-17:
@bigtreetech Any info please ?

whitedavil, 2020-06-19:
Good morning.
I have already buy this new TFT50 screen. Can you upload the firmware for customization? 
Thanks

Guilouz, 2020-06-21:
No reply on GiHub, Facebook, mail and Aliexpress MP where I buy it... It's not very serious... @bigtreetech

Tygrys-1, 2020-06-21:
I think they are working on the sale now. After the sale ends, things should get back to normal. As the software will be probably the same, you may try to use config.ini to customize. As the icons for TFT70 are already in the repository, I think that they will update the variants and the rest of the stuff here.

However, I'll not expect bells and whistles from the new boards. They seam to be just bigger LCD's in size. Just my thougts based on the pictures and advertising.

bigtreetech, 2020-06-30:
The firmware of TFT43/50/70 will soon be merged into this repository

max5962, 2020-06-30:
> The firmware of TFT43/50/70 will soon be merged into this repository
Thanks !! Compatible with tft35 e3 screen ?

luc-github, 2020-06-30:
@bigtreetech any timeframe ? I need to see why ESP3D freeze the TFT when sending `M117 hi `
and why some commands like `MD SD:` `M20 U:` `M115 TFT` are ignored
Thank you

Guilouz, 2020-07-01:
> The firmware of TFT43/50/70 will soon be merged into this repository

Soon it means everything and nothing... I receive my TFT43 in few days and I would like to be able to use it!

bigtreetech, 2020-07-02:
@luc-github hello,  It will be uploaded after cleaning up today

oldman4U, 2020-07-02:
@Guilouz 

This software is Open Source, so it is made by unpaid members of this community. It would be great if you could think about how YOU can help to make this work (or to improve it), until you get your new display.

Thank you

luc-github, 2020-07-02:
@bigtreetech thank you, seems issue was with FW provided with screen - no problem with latest precompiled neither sources

haplopeart, 2020-07-17:
I just got my TFT 70 and tried to flash the screen.  It seems to start, but ends up at iLLegal flash App.  Now I am stuck there.

I used a 8GB SD (same one I use for my TFT 3.5 E3).  To be clear I wiped out the files for the TFT3.5 first, I loaded the new files for the TFT 70.

Any assistance or way back to a working screen would be great.

oldman4U, 2020-07-17:
Hi. Have you tried the reset.txt like described?

haplopeart, 2020-07-17:
reset.txt is a calibration reset to my knowledge.  It has nothing to do with a failed flash.  I tried it none-the-less...No Good

oldman4U, 2020-07-17:
Maybe you can attach a picture so we can see where you stuck.

closed 2020-07-28 by bigtreetech

Vyacheslav23, 2022-01-16:
#Почему экран BigTreeTech_TFT70_V3.0 перестает работать когда в разъем WIFI на экране ставиться модуль ESP01 с  работающим на другом принтере проект WIFI ESP3D?  Затыкается, пишет нет связи. Как исправить эту проблему?

Vyacheslav23, 2022-01-16:
Почему экран BigTreeTech_TFT70_V3.0 перестает работать когда в разъем WIFI на экране ставиться модуль ESP01 с  работающим на другом принтере проект WIFI ESP3D?  Затыкается, пишет нет связи. Как исправить эту проблему?