bigtreetech/BIGTREETECH-TouchScreenFirmware #799: [BUG] (short description)

Chukozor, 2020-06-16:
### Description

### Steps to reproduce

1. update the firmware of my TFT35 3.0 with the already compiled firmware in the folder "copy to SD card root directory to update - unified menu material theme " of the last commit 2a47f6f1db865d0c67d191d6ab95a2f987bb9f8 (friday 2020/06/12)

**Expected behavior**
I expect the printer to adjust the height when printing as the G29 told it.

**Actual behavior**
At each boot the message "Fade height off" is appearing, **even after a G29 and M500 and a reboot**.
I use BLTouch v3.
I printed 5 squares at each corner of the bed and on at the center : the one at the center is fine (I use BLtouch to Z-Home) but the 4 others are not printed well, some are printed from too high from the bed and some are printed from too low.
The fade height was still working fine with the version 0.25.1 I used before (I'm back to this version to print well).

### Hardware Variant

TFT35 3.0, SKR 1.3 board
### TFT Firmware Version & Main Board Firmware details
TFT35 firmware of commit 2a47f6f1db865d0c67d191d6ab95a2f987bb9f8 in the forlder "copy to SD card root directory to update - unified menu material theme"
SKR 1.3 board with the marlin version on BTT github for SKR 1.3

Tygrys-1, 2020-06-21:
It is showing a message about "Fade height 10" on my TFT in 26.1, so I think that it is actually working. Maybe you should check the auto_load_leveling:1 in config.ini. Or, if you are using the "Start gcode" option from the menu, to set the restoration of bed leveling and fade height there.

After upgrading the version it is a good practise to reset the configuration settings from the menu, as they might change the defaults or the parameter storage format.

If all else fails, you may add the restoring of the leveling in the gcode in the slicer settings.

Chukozor, 2020-06-22:
@Tygrys-1 Thank you for your answer. I'm gonna try it.

Hipsta69, 2020-06-22:
I am having the same issue.
My auto_load_levelling:1 is set. Also if I go to the custom menu and choose "Restore Levelling" it turns off again straight away

Tygrys-1, 2020-06-22:
Restore Leveling sends M420 S1 gcode, and depending on your leveling method it should enable the mesh. But, any G28 afterwards will disable it again (unless you compile the Marlin with an option to not do so). Of course you have to have a valid mesh stored in eeprom before using M420.

Hipsta69, 2020-06-23:
I use a BLTouch, with the mesh saved to EEprom.
In marlin, I have #define RESTORE_LEVELING_AFTER_G28 enabled, this is also enabled   #define ENABLE_LEVELING_FADE_HEIGHT

The only way I can turn the fade height message off is by running the ABL command (in the touch screen)

Tygrys-1, 2020-06-23:
Do you print from printer's SD or form TFT?
If TFT - do you use "Start gcode"?
If you use it - how does it look like?

Do you set Fade Height in Marlin (and saved to eeprom) - to what value?

Hipsta69, 2020-06-23:
Sometimes from the TFT, other times Octoprint
I don't have the 'Start Gcode' turned on.

I have the G29 command in the start Gcode of Cura but the printer ignores it if I don't run ABL when I turn the printer on.

I cant actually find a setting in marlin (2.0.5.3) to define a height

Tygrys-1, 2020-06-23:
The Fade Height can be set in Marlin menu, somewhere in the Bed Leveling section. There is no entry for default in configuration. 

To add Fade Height to your print gcode use [M420 Z](https://marlinfw.org/docs/gcode/M420.html)

BTW, I had Bilinear Leveling before, now switched to Universal (UBL) and this seams to be more acurate and faster.

idjmic, 2020-08-02:
> The Fade Height can be set in Marlin menu, somewhere in the Bed Leveling section. There is no entry for default in configuration.
>
> To add Fade Height to your print gcode use [M420 [...]

Is M420 Z in start code in replace of M420 S1 or with?

closed 2020-08-19 by bigtreetech