bigtreetech/BIGTREETECH-TouchScreenFirmware #801: Banner text in Marlin Mode and Custom G-Code Commands

AntoszHUN, 2020-06-16:
In Marlin mode, it does not print the #define ST7920_BANNER_TEXT text I entered. I would ask for your correction. Once it was good. Then why did you have to tame this too? They always touch what they shouldn't. Another problem. Custom G-code Commands do not change when I rewrite. That was good before, too. Why had to be disturbed, varied, changed?

guruathwal, 2020-06-16:
did you use config.ini file to change those settings?

AntoszHUN, 2020-06-16:
If everything is there, what's the meaning of the configuration.h file? So far it has been as good as it was. Slowly, for a few settings, 5 files need to be edited. It will not be enough to just rewrite configuration.h. Instead of being simpler, it complicates everyone ...

Tygrys-1, 2020-06-16:
configuration.h is for setting firmware defaults.
config.ini if for setting user defaults.
This is good, because to change user settings you do not need to install the programming enviroment, know about platform.io, code editing, compiling, and so. You use any text editor and set your own configuration at will.

If you do not like the config.ini file just do not use it at all. Do your changes in configuration.h, rebuild, upload, go to features, reset config and voila!

AntoszHUN, 2020-06-17:
Super. How can I not use it? How to exclude? I find it completely unnecessary. You don't have to constantly adjust the firmware to need an ini file, so you don't need to install a firmware for a few settings. I think it's unnecessary. But it doesn't matter. This is. There will never be a final finished version of this firmware either. You always have to change something about it ... But why? ...

guruathwal, 2020-06-17:
@AntoszHUN 
No adjustment in firmware is needed to use an **ini** file. **config.ini** is not a part of the firmware source code.
If you do not want to use it, just do not look at it. The **configuration.h** works the same ways as it did earlier.
If you know how to compile the firmware, you also know that in order to load any changes from the configuration.h you need to reset the settings from the Feature settings menu or by creating a **reset.txt** file in SD card.
Even before the introduction of **config.ini** the procedure was the same. 

As @Tygrys-1  has explained very well
By adding a config file there is no need to compile the firmware every time to change a few settings or to upgrade the firmware.
With a config file all you have to do is:
**Edit config.ini** > **copy config.ini in SD card** > **press reset button**
To reset settings to original defaults just go to the Feature settings menu and reset.

with configuration.h
**Edit configuration.h** > **Compile Firmware** > **copy firmware to SD card** > **press reset button**
To reset settings to original defaults do the same procedure again.

There is never a final version of any software/firmware which is under development.

AntoszHUN, 2020-06-17:
Of course I am aware of that. I just didn't get the point. But now it's clear. Thanks for the reply! I close the topic.

closed 2020-06-17 by AntoszHUN