bigtreetech/BIGTREETECH-TouchScreenFirmware #804: [FR] Transfer a file via Wifi ESP3D to the TFT-SD and/or TFT USB-Stick

Wesie, 2020-06-20:
Hi, 
I missing the possibility to transfer a G-code file via Wifi (if the ESP3D connected) from PC to the internal TFT SD-Card and/or TFT USB Stick. 
In this moment the SD-Card or the USB Stick must be removed, connect to the PC, walk back to the TFT, insert the SD-Card and/or USB Device to continue. 
It will be much easier, also if the TFT covered in a Box, if the filetransfer works via Wifi.

oldman4U, 2020-06-20:
Hello Wesie.

This should be possible using the web based client software. How do you use the WIFI function today?

Wesie, 2020-06-21:
Which SW as Web Client does you used?
I've check the code and did not see any command to a) create a subfolder b) create a file in the TFT SD Card or TFT USB Device. 
Background : GRBL does not have a SD Card or USB device on theachone controller. So it's necessary to use the memory save option on the tft display

oldman4U, 2020-06-21:
I use the standard ESP3D interface, which allows me to see the content of the SD/USB Stick attached to the TFT and also to upload new files. Not sure there is a way to generate subfolders or new files, but the headline of your tickets asks for something which works on my machine.
[image]

Wesie, 2020-06-21:
Only for sure:
You can create a subfolder on the SD Card which is insert in the TFT Display?
You can transfer a file from PC to the SD Card which is insert in the TFT Display?
Or does you mean (that's is what I believe) you can transfer files from PC via WiFi over the TFT zu the SD Card of your control board (e. g. Marlin)

oldman4U, 2020-06-21:
I can

Transfer a file via Wifi ESP3D to the TFT-SD and/or TFT USB-Stick

zozo6015, 2020-06-21:
How did you solved the 8.3 filename issues? I am trying to upload files but I always get this error? I have long filenames enabled in marlin but it does not seam to work with ESP3D.

oldman4U, 2020-06-21:
This is another topic, but as far as i know, using the ESP3D you have to use 8.3 file format and that there is currently no work around (solution) for this.

BTW. Even Lic helped me a lot, i was not able to upload the newest version of his firmware/software, so i stuck with this: https://github.com/luc-github/ESP3D-WEBUI/issues/83

Replaced Printrun for me and does what i need.

Wesie, 2020-06-21:
Can you make a small test please? 
Power only the Display with the wifi ESP3D board. (without the 3D printer controller) 
Make a connection from the PC to the display and try only to create a subfolder. 
I didn't see any code to create folder and/or files in the Display source code. There is code available for list the directories, go into the subfolder, print the file and delete the file. 
May I haven't see the right part... but I can not create a subfolder on the TFT-SD or TFT USB. I can upload files into the internal flash of the ESP3D... that's working..

Tygrys-1, 2020-06-22:
Can you create a folder on printer's SD card from Marlin interface or using the remote terminal?
What GCode is used for this?

oldman4U, 2020-06-22:
Hi Pussycat;-)

Are you asking me or Wesie?

Tygrys-1, 2020-06-22:
Hi, the Aged One ;-)

The question is for Wesie (mainly), but the concept of creating folders on SD is new to me and I am curious, how to do it from GCode, as I have never seen such an option. So whoever answer how to do it will be ok. 

I'm not considering transfering files through the WiFi, as the WiFi connection to the printer/TFT is done through the RS-232 serial protocol, and it is not efficient enough.

Ability to use a WiFi dongle in the USB port might be interesting if someone want to remote the file transfer to the printer.

oldman4U, 2020-06-22:
Olden but golden, kitty!!!

I believe Wesie knows much more about the ESP stuff than you and me together, so maybe this is just a test;-)

@Wesie. I finally gave up the same time wie ich bemerkt habe, dass es keine Verbindung zwischen der Slicersoftware und der ESP Lösung gibt. Damit ist jeder Workflow 10x komplizierter als ein USB Stick. Die Steuerung und das Überwachen benutze ich allerdings sehr oft. BTT hat ja jetzt die "Cloud" Lösung herausgebracht, vielleicht versuche ich die einmal.

What do you say little cat, do you agree?:-)

Tygrys-1, 2020-06-22:
For wireless printint through ESP you may take a look at https://github.com/probonopd/WirelessPrinting

Standard ESP interface enables (along with the web interface) the terminal access to the printer on port 8888. So, if you can tell the slicer the ip adress and the port, it will be able to controll the printer through the GCode.

I do not like the idea of the print thorough the net, as any communication glitch could ruin the print.
For print monitoring, I prefer the camera ;-)

I know that I know nothing :-)

Wesie, 2020-06-22:
Zitat: >> so maybe this is just a test;-) <<
It isn't a test .... 
I have a small CNC 3018 with a woodpecker board, which contains Grbl as OS (No Marlin or Smoothware.... ). This board does not supported an internal SD Card. The default remote offline controller is  for trash after I've installed optical endstopps . That's the reason, that I've buy the BTT 35 V3 Display with the ESP3D. In this moment I try to modify the code, that the Display is working with Grbl. 
About all the things above, it will be very nice to have the possibilty to transfer a file via Wifi to the Display SD / USB from PC after the gcode file is finish.
If you look on the headline: It's a feature request. So I was surprised, that this one should be worked ... Ok, I can make also some mistakes, which it let this feature no more working anymore. But I've double check them. 
If a feature request closed with the last meaning (additional for all other, which are searching  this item in feature) that all things are working: The Feature request will not be pulled and not included. Add. further user are confused, because they believe, that there is another mistake.
I know the Web-Gui for the ESP3D. I work also with the same GUI.
In the meantime I've read a comment from the BTT dev (?) that there is a possiblity to write files on the TFT SD / TFT USB in future, if there are more people want this one and make a request for this feature.

Wesie, 2020-06-22:
> For wireless printint through ESP you may take a look at .....
> I do not like the idea of the print thorough the net, as any communication glitch could ruin the print.

I'm totally with you. ButI don't want have a PC all time beside. A Notebook goes into Sleep-Mode .... its sometimes not a good idea to have expansive devices beside Milling / Printer ...
Gcode-files are not the biggest one: Transfer to the SD / USB and let print them from the Display alone ;)

oldman4U, 2020-06-22:
Hi Tiger

I would ever print over network just copy the files over. But to save them
to the storage, open another app to select the file and copy it over - no.
Save to printer storage directly from Cura would be nice and also the
ability to copy firmware from os to the printer.

Tygrys-1, 2020-06-22:
In that scenario (GRBL board) you are 100% right - the CNC lacks the built-in SD on default boards. The Idea of using the TFT is ok, but keep in mind, that it is designed mainly for 3D printing, initially as a terminal to the Marlin and most features are directed on this usage scenario. The communication language is GCode and ESP WiFi is mainly just redirected to the Marlin's serial input. The fact, that it can interact with TFT part of the board is a hard work of Luc, the author of ESP3D project. I think, that functions you request requires both ESP3D and BTT TFT code change, so you do better ask Luc about this. You can use ESP3D to transfer files to BTT SD or USB, maybe you have to use 2.1 BTT TFT branch to have these options. I have not heard of the way to make a new folder through the GCode.

Tygrys-1, 2020-06-22:
Hi, oldman4U,
There is a problem with a way that M28 works, and this makes file transfer to the printer's SD SLOW. For me it definitly is faster to go to the printer, take the SD, insert in my computer, copy files to print, go back to the printer and insert the SD there. But I have my printer in the next room, so not a big deal. 
I'll try to test transfer to the TFT's SD or USB and maybe it will be limited only by the ESP baud rate setting.

Wesie, 2020-06-22:
> The fact, that it can interact with TFT part of the board is a hard work of Luc, the author of ESP3D project

Luc has prepare the WEB-Gui for this feature. Uncomfortable the last code in the BTT Dispaly is missing :(

> keep in mind, that it is designed mainly for 3D printing, initially as a terminal to the Marlin and most features are directed on this usage scenario. The communication language is GCode and ESP WiFi is mainly just redirected to the Marlin's serial input.

I would say: round about 70% is working :)   :)  :)  My problem is more or less, that I need more expierence of Milling.. Some issues must be solved on the TFT,too.

Tygrys-1, 2020-06-23:
> 
> I can Transfer a file via Wifi ESP3D to the TFT-SD and/or TFT USB-Stick 

If I swich (Web Interface) the SD list to the TFT SD or USB, the upload button vanishes.

oldman4U, 2020-06-23:
Hi Tiger.

You are maybe right and i had this wrong in my head, sorry. But the latest and greatest software from Luc should be able to upload also to the TFT and Wesie commented in the corresponding ticket 2 days ago: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648.

So Wesie should also be aware.

Wesie, 2020-06-23:
let's cross the fingers, that the missing feature will be included in the next near future :)
Gcode files are not so big and it will be also help for modification of the code, if the firmware could be transfered via Wifi. I have the feeling, that the SD Card slot have with the time more and more place (The card goes more smoothly in and out with the time.... hopefully the contacts (Reader to card) are still available if I'm finish..

oldman4U, 2020-10-20:
Hello Wesie.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

Wesie, 2020-10-20:
close on request

closed 2020-10-20 by Wesie

4kardes, 2022-11-14:
> […](#)

How? which repository do you use