bigtreetech/BIGTREETECH-TouchScreenFirmware #806: [BUG] (short description)

dmiart, 2020-06-21:
### Description

TFT24v1.1 becomes unresponsive when GCODE has a pause at layer# and then you go in TFT and press on extrude button. After that its no longer possible to press any buttons on the LCD. I am using the latest release 

### Steps to reproduce

1. Cura set pause at layer # 
2. print file from sd card on TFT LCD
3. wait for printer to pause at specified layer
4. go into menu to load/unload fillament on TFT LCD. 

**Expected behavior**

Expect to be able to unload fillament

**Actual behavior**

LCD becomes unresponsive

### Hardware Variant

TFT24v1.1
### TFT Firmware Version & Main Board Firmware details

Vx.x.26 firmware on TFT24v1.1 & Marlin 2.0.5.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Tygrys-1, 2020-06-21:
Take a look at  #758, #794, #760, #773

bigtreetech, 2020-09-01:
Hello HOST_ACTION_COMMANDS should be worked after #984

closed 2020-09-01 by bigtreetech