bigtreetech/BIGTREETECH-TouchScreenFirmware #807: [BUG] (short description)

spencerandhill, 2020-06-24:
### Description
When I boot up my SKR 1.3 Controller with the TFT35 V2.0 Screen (Touch-Mode) my Z-AXIS always moves up "some" height. It's definetely more than 10 or 20, feels like it's traveling the Z-Value of "DEFAULT_AXIS_STEPS_PER_UNIT" in Marlin 2.0

Without the TouchScreen, there's **no raise** of the Z-AXIS.

The controller and touchscreen is used for a mpcnc.
The behaviour is also described here: https://forum.v1engineering.com/t/tft35-funky-boot-up/18117/7

I know, that there's a parameter "NOZZLE_PAUSE_Z_RAISE" and "POWER_LOSS_ZRAISE" but they have no effect, even when I put them to 0.

**Expected behavior**
I don't want the TouchScreen to move my Z-AXIS on bootup, or at least I would like to turn that off by setting "POWER_LOSS_ZRAISE" to 0.

**Actual behavior**
It moves the Z-AXIS off everytime, no matter what I do. Even, when I shutdown the controller in the menu and then reboot the screen/controller, it still moves the Z-AXIS.

### Hardware Variant
SKR 1.3 ( LPC1768 ) together with TMC2208 drivers
TFT35 V2.0 connected by UART

### TFT Firmware Version & Main Board Firmware details
Board: Marlin 2.0
TFT35 : V2.0.25

oldman4U, 2020-06-25:
Hi two persons in one.

Could you please update to the latest version of the firmware and attach the config file after that.

Since when do you experience this issue and what changed at that time?

oldman4U, 2020-06-26:
hi.

any news regarding the reported issue?

spencerandhill, 2020-06-28:
Hi oldman4U,
unfortunately i'm not at home until monday, so I can't check that. I'll check as soon as possible and report, if this fixed the issue.

oldman4U, 2020-07-03:
Na dann lass es uns wissen, wenn es etwas neues gibt.

Danke

spencerandhill, 2020-07-07:
Hi!
I tried to flash the new firmware to the TFT-Board, but unfortunately I get a very distorted and squishy screen. I can't even see any icons, except the calibration-procedure. Does anybody know, why this happens? I can't check, if the z-axis is moving or not, cause my screen isn't working with the new firmware.

Reflashing the old firmware, makes the screen working, but then I got the other issue back :/

oldman4U, 2020-07-07:
Hi.

Flashing the new firmware you should have 3 things on the SD card.

1. the firmware
2. config.ini
3. TFT35 folder with font and bmp in it

Try it again and send a picture in case it does not work. Also error
messages during the update shown on screen could help.

spencerandhill, 2020-07-13:
Hi! I just flashed the updated firmware today, but the issue still existed. Therefore I looked in the marlin code again and found a parameter in Configuration_adv.h which said "#define STARTUP_COMMANDS "G1 Z40""

Well guess what? This was the issue. Therefore, there's no issue with the tft display, but it was in my Marlin firmware.

This issue can be closed. Thank you for your help and patience @oldman4U !

closed 2020-07-13 by spencerandhill

oldman4U, 2020-07-14:
👍

You are very welcome