bigtreetech/BIGTREETECH-TouchScreenFirmware #808: [FR] (knob_led_color idle delay)

devildant, 2020-06-26:
**Is your feature request related to a problem? Please describe.**
when the idle delay of the screen activates the knob led stays on.

**Describe the solution you'd like**
I would like there to be an option so that when the screen goes to idle, the knob leds goes off, and it turns back on with the same color when the screen comes out of idle

**Describe alternatives you've considered**

**Additional context**

traffic-light, 2020-08-05:
Oeh. That sounds nice. Currently I'm working on it. When I'm ready (or something else maybe) with it I'll post an update.

traffic-light, 2020-08-05:
At the moment i'm planning to make it work together with the LCD screen. When the LCD go's idle the knob's LED will go out (or will dim if possible. (it should be possible because it's a neopixel)

### Updates

- I've added the menu item and settings management. [image]

To do:

- [x] Add menu item to the featuremenu (toggle idle mode)
- [ ] Make the button go off when the screen go's in idle
- [ ] Make it possible to set a idle precentage

devildant, 2020-08-05:
nice, it's progressing well :)

traffic-light, 2020-08-05:
It's almost done. The only thing that I can't get working is the settings.ini
It's telling me over and over again that ''''knob_led_idle'''' keyword is not found...
So.. I'll have to dig a little deeper for that one.

Maby someone around here knows where I have to add new keywords? 
I've placed it in *flashstorage.c*
````

  wordToByte(infoSettings.knob_led_idle,              data + (index += 4))
````

traffic-light, 2020-08-05:
I've made a pull request. Unfortunatly it's not compiling on **BIGTREE_TFT35_V3_0**

If you wan't to test it, see #914 
(or my fork https://github.com/traffic-light/BIGTREETECH-TouchScreenFirmware)

closed 2020-08-06 by bigtreetech

devildant, 2020-08-06:
Thx @traffic-light I will test it as soon as possible (I saw that it was merged, nice)

devildant, 2020-08-06:
hello, i test, 
this crashes the LCD emulation :(
[image]

traffic-light, 2020-08-06:
Oh... So my code do's that? I'm sorry. I'll try to fix it

traffic-light, 2020-08-06:
See my bug in issues. It could be indeed a problem with my code. I'm searching for a solution

devildant, 2020-08-06:
do not worry :)
I saw the problem because I only use the marlin mode ^^

traffic-light, 2020-08-06:
It's happend whit the other commits. the version on my git works flawless for me

devildant, 2020-08-06:
arf, if i disable knob idle no pb when screen idle, but if knob idle is activate, when the screen idle, marlin mode crash

traffic-light, 2020-08-06:
Yup. It's also in the version on my GIT. I'm trying to fix it.

devildant, 2020-08-06:
in my opinion it smells of electromagnetic interference.
can you reduce the brightness to the maximum to see?
I also note that when you switch the knob to off from the touch screen, this does not crash the marlin mode, you should see the actions that are done from the menu

traffic-light, 2020-08-06:
yes. it could be indeed....
I will take a look at it. Thank you

devildant, 2020-08-06:
no problem, I was confronted with this kind of anomaly on personal projects

traffic-light, 2020-08-06:
Yeah. me to. One time a motorcable was interfering with the encoder cable at work. The motor did the most strange things. Go up with the speed of sound into the endstop :stuck_out_tongue_closed_eyes:

traffic-light, 2020-08-06:
I've found the problem..
You can download the new working version here:
https://github.com/traffic-light/BIGTREETECH-TouchScreenFirmware/commit/b4f4c1c6b7aa47cda4c729800fa4b7c6b0fbdd13

Can you please test it for me (including updating font, icons and the config.ini

(i also added a 0% screen brightness option. off is 0 and 11 is 100% in the config. Maybe I take a look if it's possible to do -1 for 0%)

devildant, 2020-08-06:
nice, arf currently, I launched a 2 day impression, I will test as soon as possible.
it's __disable_irq(); and __enable_irq(); who was causing the problem ?

traffic-light, 2020-08-06:
Yes, that was the problem. Originally it wat around the whole for loop as you can see.
When you switch it off to long it's getting messy.

The only strange thing is that there was no problem with switching it on...

devildant, 2020-08-07:
hooo sorry i write impression (in french :p), 2 days printing :p

traffic-light, 2020-08-07:
oh. haha. nice. That's a long print.

devildant, 2020-08-24:
well I have re-tested (sorry for the response time I was too busy: p)

everything is working fine now nice job

traffic-light, 2020-08-25:
No problem. Good to hear

PR #914 (2020-08-06): Added idle to the rotaryencoder knob led
You can now select if the led of the rotaryencoder knob should be turned off when the LCD is in idle

### Related Issues

Function request: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/808

So, pull request resolves #808 

I'm planning to set the idle percentage. This way you can select what the idle brightness is.