bigtreetech/BIGTREETECH-TouchScreenFirmware #809: [BUG] TFT35 E3 V3 Knob axis is metall :O

Wesie, 2020-06-26:
It's not a mistake, its not Works as designed... its a bug

Today I received the TFT35 E3 V3 Display.
Surprise for me: The LED are under the knob (Plastik Cover) but the axis from the encoder is from metal *upps*

In a normal case the axis is from transparent plastic, so that the PCB can be covered without any issues... In this case I must leave (how many???) mm space for the light - if a small procent cames out of the hole and could passed the knob
It's a no go .....

Tygrys-1, 2020-06-27:
It is called E3, because it is designed for Ender 3, where it fits and lights as it should. The Ender 3 case hole for encoder is 10 mm diameter.

Wesie, 2020-06-27:
It's a valid argument, 
on other side, it will be much better - also for user, which want to use the Display in an own project, if the axis will be transparent and the light goes through the axis to the knob.

closed 2020-06-27 by Wesie