bigtreetech/BIGTREETECH-TouchScreenFirmware #810: [FR] Octoprint Print Status Screen

craigistired, 2020-06-28:
When printing from Octoprint there is no option on TFT35 V2.0 to display the print status screen to see the progress bar (M73/M78) or filename (M33).

I would like to be able to access the print screen even when not printing from an SD card.

Hipsta69, 2020-06-28:
the 'Detailed Progress ' plugin sends M117 messages to the screen, thats what I use

craigistired, 2020-06-29:
@Hipsta69 Are you using the Unified Menu Theme?

Hipsta69, 2020-06-29:
Yes, I'm using the Unified menu. I also have Octopi connected via UART4 on the screen so I don't know if that makes a difference?

craigistired, 2020-06-29:
@Hipsta69 Could you post a screenshot? Where does it display this info, on the main menu?

Hipsta69, 2020-06-29:
@craigistired I have a video of it in action on my youtube here: https://youtu.be/1pZ7j14yweQ

Note that in the video the display is set to scroll every 2 seconds, I did this just for the video and the default is 10 seconds but this is easily changed in the plugin

craigistired, 2020-06-29:
@Hipsta69 That's interesting. Even with the "Detailed Progress" plugin it only displays "(i) Info" and "Ready". I'm connected via USART1 (RS232).

kind3r, 2020-07-01:
@craigistired you would have to connect your PI to the TFT via one of it's UARTS.
There is also an older thread for this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281

ibrahimcengiz, 2020-08-17:
I have same problem, actually i need babystepping control when i start a print from octoprint.

oldman4U, 2020-09-03:
Hi @craigistired .

What is the status of the reported issue.

Please close this ticket in case you do not need it anymore

Thank you

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

craigistired, 2020-10-20:
"Your feature request is covered in the pinned Feature Request ticket #1177"

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1177

closed 2020-10-20 by craigistired

oldman4U, 2020-10-20:
Typo, it is Ticket 1170 and not 1177.

Sorry

Thank you for closing it anyway;-)

oldman4U, 2020-10-21:
Wanted to add, that more M117 support and fixes are coming with #819 
Maybe you can check and see if this PR solves the issue.

Thank you

DaStivi, 2021-03-02:
i would like to raise this FR too... have setup octoprint right now... also enabled all the progress display plugins, at its quite ok, to see atleast something...  

but it would be very nice if there is at least some basic "print" display shown when an printjob from the pi has started... you can even use the M117 commands for the progress etc... and it might even be possible to add the basic function buttons as, temp change, fan speed etc... everthing that didn't have to tinker with the gcode should be fine i guess...