bigtreetech/BIGTREETECH-TouchScreenFirmware #813: Changing the temp while heating up for a print is ignored

oldman4U, 2020-07-02:
until the Temperature has been reached

### Steps to reproduce
Start a print from TFT USB
While heating up the bed or the nozzle, change the temp of the element which is heating at that time
**Expected behavior**
I would expect that if I lower the temperature, the heating stops at that temp, like it worked in previous versions

**Actual behavior**
It heats up to the temp defined in gcode

### Hardware Variant
TFT35 E3 v3
### TFT Firmware Version & Main Board Firmware details
Latest version from today 2.7.2020

luc-github, 2020-07-02:
@oldman4U I think it is Marlin behavior
how did you start heating ? depending of heat command :` waiting until reach temp` or `no waiting`, Marlin does not allow any command when heating 
M104 / M140  (not waiting) allow to change temp and send any gcode 
M109  / M190 (waiting until reach) does not allow any GCODE until temperature is reached - you will get `echo:busy: processing` as answer

oldman4U, 2020-07-02:
Hi Luc.

Thank you for your answer.

I started the print by selecting the gcode on the USB Stick, so without preheat. The file is attached.

In the meantime I downgraded to a TFT version from June, and there I can not see this behaviour. There the bed stops heating at 40 for example, when I change it from 60 to 40 before the bed reaches 40 degrees.

[CE3_Spare.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4865873/CE3_Spare.gcode.zip)

oldman4U, 2020-07-02:
Just tested the build from yesterday and there it also works correctly.

luc-github, 2020-07-02:
so you mean no more issue?

oldman4U, 2020-07-02:
Using a version up and including yesterday, July 1st, everything is ok.

Using the version from today, July 2nd, the reported bug exists.

This is reproducible by changing the firmware only. Everything else remains the same. Printer hard and software Marlin Version from June 26th, gcode file - everything.

So the reported issue has to be related to the version uploaded today.

bigtreetech, 2020-07-03:
Confirm this bug, Fix it now

oldman4U, 2020-07-03:
Thank you. The changes made by Gurmeet are great so i hope this fix comes soon. 

Thank you

bigtreetech, 2020-07-03:
Hello,  It should be fixed in #818

oldman4U, 2020-07-03:
Hi.

Thank you. I will download and test asap.

oldman4U, 2020-07-03:
Hi.

Thank you, problem solve, so I will close this ticket.

@bigtreetech
A little bit off topic. I tried to close unneeded tickets in some of the repositories (repos of hardware which I also own) by asking the owners to close them. Some did it, some not. Those which are not reacting any more I made a notice to you bigtreetech and asked to close it. Unfortunately nobody closed it so far.

I believe the ticket system will become unusable soon and we saw some tickets related to the same topic, because nobody will search several pages to see if the issue has been reported before. Maybe you can take a few minutes to close those tickets.

Kind regards

bigtreetech, 2020-07-04:
Thank you， I will review this issues  and close that unneede.

oldman4U, 2020-07-04:
This would be great. I tried to notify you only for tickets where the user
do not react and which are not important any more because they are fixed in
the meantime or not related.

But please review and let me know in case i made a mistake

Thank you

oldman4U, 2020-08-02:
Seems this bug comes back once per month. Release tested from August 2nd.

oldman4U, 2020-08-03:
Strange. Updated to the firmware from August 3rd and it works again....

Will keep an eye on it.

closed 2020-08-03 by oldman4U