bigtreetech/BIGTREETECH-TouchScreenFirmware #817: Change from unified menu to marlin menu stuck display[BUG] (short description)

Davidoff77test, 2020-07-03:
### Description

### Steps to reproduce

1. flash last fw
2. by long tap change to marlin menu
3. display is black and stucked. Also after restart

**Expected behavior**
Changing works

**Actual behavior**
Changing doesn't work

### Hardware Variant
TFT35_V3.0_E3
### TFT Firmware Version & Main Board Firmware details
.26.x / [bugfix-2.0.x]

### Additional Information
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4868551/config.zip)

oldman4U, 2020-07-03:
Hi Davidoff77test.

Could you please try
1. MASTER branch instead of 26.
2. Rotate the encoder after switching from touchscreen to Marlin mode.

Let us know

oldman4U, 2020-07-05:
Hi. Any news regarding the reported issue? Please let us know

Davidoff77test, 2020-07-06:
After rotate the encoder is the screen shows, but I need to rotate encoder after every switch to Marlin. Is it feature?

oldman4U, 2020-07-06:
It is the intended way it works, because it does not get an update from Marlin. Please see here for more information:

https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/blob/master/Hardware/Explanation%20of%20black%20screen%20in%20TFT35%20V3.0.pdf

Please do not forget to close this ticket once you do not need it anymore.

Thank you

kind3r, 2020-07-06:
Indeed Marlin needs to update the display for it to show on the TFT screen. You can trigger the update  by rotating or pushing the knob or it should refresh automatically if you are on the status screen and the information changes (temps, fan speed etc.).

closed 2020-07-06 by Davidoff77test