bigtreetech/BIGTREETECH-TouchScreenFirmware #826: [BUG] Encoder glitch when adjusting fan speed, speed and flow

kind3r, 2020-07-07:
### Description

When adjusting fan speed, speed and flow using the encoder in TFT mode the respective value glitches into a loop going up and down in value until reboot.

### Steps to reproduce

Rotate the encoder knob fast enough when in fan, speed and flow menu.  
Made a short video about the issue: https://youtu.be/u0Gz-jvzVVs

**Expected behavior**

The value should change up and down depending on the rotation.

**Actual behavior**

The value is stuck in a glitch automatically changing up and down by a few increments until reboot

### Hardware Variant

TFT 35 V3.0

### TFT Firmware Version & Main Board Firmware details

Current firmware built from source (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/868d61549cc480a06c6c2526b828efab5807539a).

### Additional Information

The issue is not configuration dependent and was introduced in the https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/41a11faf2384957114bb23421895fa861b6458d4 commit.

kind3r, 2020-07-07:
Finally discovered it's caused by the duplicate updating of values by the encoder and by the command queue.
Will put in a PR.

oldman4U, 2020-07-07:
Would be great if this could be fixed. Happens also when using the touchscreen and not the encoder and stops the print

kind3r, 2020-07-08:
> Would be great if this could be fixed. Happens also when using the touchscreen and not the encoder and stops the print

I had a feeling that would be the case but I could not reproduce it by only using the screen. Anyway, you can try my PR and check. For me it's working so far.

digant73, 2020-07-08:
> Would be great if this could be fixed. Happens also when using the touchscreen and not the encoder and stops the print

yes, I confirm that it happens also to me with the touchscreen. I also see that changes on speed and flow are not reset to 100% after a print. They will remain also for other prints. A reset is needed to restore default values.

kind3r, 2020-07-08:
> I also see that changes on speed and flow are not reset to 100% after a print. They will remain also for other prints. A reset is needed to restore default values.

This is not the case for everyone. You can reset them in your start or end g-code of your slicer.

gordo3di, 2020-07-08:
We had the same issue with speed and flow when just adjusting values using the + - buttons. Crashed the print. This happened on the TFT35 E3 V3 and the TFT35 V3

closed 2020-07-09 by bigtreetech

radek8, 2020-07-12:
The problem with the fan when using the coder persists, even after a heavy repair

kind3r, 2020-07-13:
@radek8 Did you compile from source ? The binaries were not updates.

oldman4U, 2020-07-13:
Maybe radek8 has two fans and we all have only one?

radek8, 2020-07-13:
Yes, I compiled a new bin file.
No, I only have one fan.
The + and - buttons work well, everything is fine when the encoder is turned slowly. But when you turn the encoder quickly, a problem occurs as in the video above

oldman4U, 2020-07-13:
I can confirm this bug;-( On my system it starts with the problem, that i can change the fan speed to spin faster, but not lower it during preheat and also right after the printer is turned on.

Sorry radek8, but i usually do not change the fan speed.

radek8, 2020-07-13:
I should have some partial solution, but it doesn't work 100%. If a problem occurs with the ticking of the value, just leave the fan menu. There is no need to reset the machine.
But such a solution is not right.

radek8, 2020-07-13:
I guess I have a solution to my problem.
I will test and publish PR.

oldman4U, 2020-07-13:
👍