bigtreetech/BIGTREETECH-TouchScreenFirmware #829: [BUG] Filament Runout Triggers Endlessly

gordo3di, 2020-07-09:
### Description

Turn on runout sensor and start a print. Pull out filament and let it trigger. After inserting filament and continuing the print the runout will keep triggering after you resume. Using latest firmware from 7-2-20

I looked at the firmware from 5-5-20 and it looks like the runout loop has changed slightly.

### Steps to reproduce

**Expected behavior**
Runout should not trigger if filament is inserted.

**Actual behavior**
Keeps triggering

### Hardware Variant
SKR v1.4 turbo, TFT35 V3

### TFT Firmware Version & Main Board Firmware details
TFT CONFIG_VERSION 20200530

### Additional Information

I fixed the issue by modifying the filament runout loop in extend.c

Change if (!getPrintRunout()) return; to if (!FIL_IsRunout()) return;

Not sure if this is correct but it solved the issue.

```
void loopFrontEndFILRunoutDetect(void)
{
  //if (!getPrintRunout()) return;
  if (!FIL_IsRunout()) return;

  if (setPrintPause(true,false))
  {
    popupReminder(DIALOG_TYPE_ERROR, textSelect(LABEL_WARNING), textSelect(LABEL_FILAMENT_RUNOUT));
  }
}
#endif
```

SulzRohr, 2020-07-15:
This solution worked for me too. Many many thanks.

drlohwk, 2020-07-18:
somehow for this method it doesn't trigger the filament runout. I need to revert back to June 12 version of 0.26.x
 that has this extend.c code.
void loopFrontEndFILRunoutDetect(void)
{
  if (!getPrintRunout()) return;

  if (setPrintPause(true,false))
  {
    setPrintRunout(false);
    popupReminder(textSelect(LABEL_WARNING), textSelect(LABEL_FILAMENT_RUNOUT));
  }
}
#endif

gordo3di, 2020-07-22:
Looks like its been resolved. Thanks.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/cce714b1c08a65781e35645dfdfbc7f0cacd1397

closed 2020-07-22 by gordo3di