bigtreetech/BIGTREETECH-TouchScreenFirmware #830: [BUG] wrong layer height (xxx.xxMM) while printing

discip, 2020-07-09:
### Description

I do not know since when this occurred, but at least it is present after **[commit 868d615](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/868d61549cc480a06c6c2526b828efab5807539a)**.

### Steps to reproduce

1. update your TFT firmware to the mentioned commit
2. start any print watch the height value
3. realize that the wrong value is shown

**Expected behavior**
The values on the display do correspond to the actual position of the nozzle.

**Actual behavior**
The values on the display do not correspond to the actual position of the nozzle.

### Hardware Variant
**MKS_32_V1_4**
### TFT Firmware Version & Main Board Firmware details
TFT|Mainboard
-|-
**[commit 868d615](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/868d61549cc480a06c6c2526b828efab5807539a)**|**[Marlin bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/commit/915847b75aeaece5d9f4ef2f2c79ec464fd30a4b)**

### Additional Information
e.g.: On the **35th layer** => the display is showing **890.48MM**. 🤨 (by the way it should say **`mm`** instead of **`MM`**)

Additionally I noticed, that the value increases with every new isle it is printing. 😵

[config files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4899514/config.files.zip)

discip, 2020-09-15:
@oldman4U 
I do not know, if anything was changed in this regard, but this issue seems to be gone.

kind regards

closed 2020-09-15 by discip

oldman4U, 2020-09-16:
Thank you and happy printing