bigtreetech/BIGTREETECH-TouchScreenFirmware #831: Firmware Install Issue Landing Screen Broken

sdharvey28, 2020-07-10:
[image]

After Installing the firmware multiple times I still get this screen. I'm worried that my screen is possibly broken but I do get the BIQU boot screen fine just the operating screen is broken. Any help is much appreciated. I'm using a BTT SKR 1.4 Turbo board if there is a bug between this screen and board

oldman4U, 2020-07-10:
Hi. 

Please put the firmware, the config.ini and the TFT35 folder which includes the bmp and font folder on the sd card and flash again. 

Let us know if this solves your problem

gordo3di, 2020-07-10:
Had the same issue a few times now with the latest firmware. I had to repload both the bmp and font folders (sometimes more than once).

sdharvey28, 2020-07-10:
Same issue. Is there a way to take it to factory settings that way  I can retry with a blank slate.

oldman4U, 2020-07-10:
@sdharvey28 

Make an empty file named reset.txt on the SD card and proceed like doing a firmware update. 

Which SD cards are you using for firmware update ( size and format)?

sdharvey28, 2020-07-10:
I bought a 64GB Sd card its file format is exFAT. Is there a way to reformat it to something else? Also the reset did not help and I tried another zip file of firmware as well and did not work.

oldman4U, 2020-07-10:
Try the SD card which came with the display as it is, or another 8 (16GB max) SD card card and format it to FAT 32 using your PC or Mac. Using such a card i was able to make 100+ firmware updates so far. In case you have the display mounted in a metal frame- remove it and try again.

Let me know

Tygrys-1, 2020-07-11:
You have to use FAT32 formated SD. Up to 32MB tested with no problem.

oldman4U, 2020-07-11:
MB>GB i guess😎

sdharvey28, 2020-07-11:
[image]

Finally got this to work with a 32GB SD card with FAT32 format. Thanks for everyone's help I was starting to go crazy.

Tygrys-1, 2020-07-11:
Your "Leveling" icon looks incorrect.

oldman4U, 2020-07-11:
I would recommend to upload the round miracle. It looks MUCH better.

oldman4U, 2020-09-03:
@sdharvey28 

You asked for help and we helped you.

Now i ask you to close this ticket, because it is solved.

It is very unfriendly to ignore us, where we helped you in our spare time

Close this ticket now.

oldman4U, 2020-09-05:
@bigtreetech 

Please close this ticket

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket

Thank you

closed 2020-10-06 by sdharvey28