bigtreetech/BIGTREETECH-TouchScreenFirmware #832: [FR] IDEA: Minor Update Iconpack 'The Round Miracle' - Update for SW v26.x

Acenotass, 2020-07-10:
Dedicated to all those who were waiting for the update of my iconpack ;)
Small update of just one preheat icon (3 icons removed)

[image]

[TFT24 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4902500/TFT24.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)
[TFT28 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4902498/TFT28.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)
[TFT35 Iconpack 'The Round Miracle' by Acenotass for v26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4902499/TFT35.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.x.zip)

labotecno, 2020-07-10:
thank you very much for your great work ! can you give me a tip how you make your iconpack ? program ?

Acenotass, 2020-07-10:
use any graphic editor, I used GIMP

labotecno, 2020-07-10:
> 
> 
> use any graphic editor, I used GIMP

great ! you make icon from something ? thank you

Acenotass, 2020-07-11:
> > use any graphic editor, I used GIMP
> 
> great ! you make icon from something ? thank you

You can see the ideas of ready-made icons, which are many on the Internet. Combine your ideas and the ideas of others to get what you want. It’s important that the general style is maintained, so that the author’s idea is visible.

gordo3di, 2020-07-11:
This is useful too. Its a tool to easily replace icons

https://github.com/bigtreetech/BigtreeDesignAssistant

Acenotass, 2020-07-11:
@gordo3di Very crude software, does not take into account the latest firmware. Suitable only to get a general impression of the appearance.

radek8, 2020-07-13:
The icons are great
Thank you

Mistigris18, 2020-07-14:
Merci,  
C'est top. Beau travail j'adore 😉👍

radek8, 2020-07-28:
Hi @Acenotass 
Will you add new icons?
global_chamber.bmp
chamber.bmp
point_5.bmp
Thank you

Acenotass, 2020-07-29:
Hi @radek8
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/885

closed 2020-11-03 by stale[bot]