bigtreetech/BIGTREETECH-TouchScreenFirmware #834: [FR] Filament change buttons

bigtreeteche3, 2020-07-11:
To make the touch setup complete i would suggest to add the filament change option like under the normal marlin setup so that people with bowden setups can load and unload with one click. 

As an alternative would it be able to mod the aktual load and unload funktion so that instead of just 10mm at the time it feeds 400 mm and more ? I can't find these line in the code...

oldman4U, 2020-07-11:
You are too late: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/769

oldman4U, 2020-09-05:
Hi.

This is covered in the pinned Read me ticket and in some other tickets.

So please close this ticket

oldman4U, 2020-10-20:
And now it is also covered in the pinned feature request ticket #1170 

PLEASE close this ticket

Thank you

oldman4U, 2020-11-21:
This feature is available in PR #1272. 

Please check

closed 2020-11-27 by bigtreetech