bigtreetech/BIGTREETECH-TouchScreenFirmware #836: [BUG] (infinity update speed, stop printing)

slawekgajzner, 2020-07-12:
### Description

Print gcode from sd-card.
When in gcode file is M106 command twice (generate from pruse slicer), TFT is hangs in infinity update speed loop.

...
M106 S206.55
M106 S178.5 
...

### Steps to reproduce

Try print this gcode from sd-card:

; generated by PrusaSlicer 2.2.0+win64 on 2020-06-14 at 10:25:38 UTC
; ...
M201 X9000 Y9000 Z500 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
;
;
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
;
;
M106 S232.05
M106 S142.8
;
;
M107 ; turn off fan 

### Hardware Variant
TFT43 and TFT50
### TFT Firmware Version & Main Board Firmware details
latest (Jul 9, 2020)

I temporary fix this problem, i added fanSetCurSpeed in parseACK.c when cmd M106 received

else if(ack_seen("M106 P"))
      {
        u8 i = ack_value();
        if (ack_seen("S"))
          fanSetSpeed(i, ack_value());
        fanSetCurSpeed(i, fanGetSpeed(i));          
      } 

Fancontrol.c
void fanSetCurSpeed(u8 i, u8 speed)
{
  curfanSpeed[i] = speed;
}

radek8, 2020-07-13:
The problem is related to the problem
#826

radek8, 2020-07-13:
same problem as #837

radek8, 2020-07-13:
So I tested it.
The new FW does not solve your problem.
If there is an M106 command in the print file twice in a row, the printer loops.
But if you merge PR #840 with a new FW, it will solve the problem.

PR #840 (2020-07-15): Fan speed repair
Correction of cycling of the fan speed value when changing speed.
Described in #826

resolves #836
closed 2020-07-15 by bigtreetech