bigtreetech/BIGTREETECH-TouchScreenFirmware #837: printer halts on any change during print

Thijsvanzwam, 2020-07-12:
### Description

When changing settings like print speed, temperature or fanspeed, the printer stops completely.

### Steps to reproduce

1. print form SD or USB
2. during any layer, change a setting like print speed or fan speed.
3. printer stops and bed and hotted stay hot

**Expected behavior**
well... that the fan speed or print speed change...

**Actual behavior**
printer stalls completely, reset is needed.

### Hardware Variant
BTT TFT35 v3.0
BTT SKR1.4, TMC2209
### TFT Firmware Version & Main Board Firmware details
latest TFT firmware and latest Marlin on SKR1.4. (tested it with older Marlin on mainboard but same issue.)

### Additional Information
The config file is the one provided by the firmware on Github, I did not make any changes. Only config.ini is changed so that is attached.
[config.ini.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4908769/config.ini.zip)

oldman4U, 2020-07-12:
Hi. Using Master TFT firmware should solve the problem.

Thijsvanzwam, 2020-07-13:
Hi,  
I have the most recent firmware grom github loaded, is that the master version?

oldman4U, 2020-07-13:
The precompiled is old. Master is new, includes the fix and has to be compiled by yourself.

Thijsvanzwam, 2020-07-13:
Ok great, I'll try that!
For some reason I can only find the precompiled ones... Would it be possible you post a link?
thanks in advance!

oldman4U, 2020-07-13:
When you go to the top of this page you can see „Code“. Click on it opens a
page which allows you to select beteween Master and x26.1. Master is the
correct one which you have to compile in VSC before you can upload it to
your TFT. Have you done this before?

Thijsvanzwam, 2020-07-13:
Yup found it, but when I select the master branch, it is precompiled.
I have uploaded it to the TFT before, and used VSC to compile Marlin for the mainboard.

Tygrys-1, 2020-07-13:
As the precompiled binaries have been updated recently, you may try current version without building your own.

To build your own you have to download or clone the repository just like with the Marlin.

Thijsvanzwam, 2020-07-13:
fixed it!
the latest precompiled binaries fixed the issue.
My own compiled one did too!

radek8, 2020-07-13:
The problem is not fixed.
Try changing the fan speed with the rotary encoder. Fast moving :-(

oldman4U, 2020-07-14:
@Thijsvanzwam 

Be aware the fan speed is still broken.

radek8, 2020-07-14:
> @Thijsvanzwam
> 
> Be aware the fan speed is still broken.

Try PR #840 works for me

oldman4U, 2020-07-14:
Aaaah. It is already available.

Great

Thijsvanzwam, 2020-07-14:
Still have the issue that no fan speed is shown during print.
fan runs but shows 0%

oldman4U, 2020-07-14:
@Thijsvanzwam 

Please open a new ticket and add all information regarding your setup especially the hardware you use and if this ever has been working properly before.

radek8, 2020-07-14:
> Still have the issue that no fan speed is shown during print.
> fan runs but shows 0%

I couldn't simulate this problem

Thijsvanzwam, 2020-07-14:
I'll try to reinstall the firmware to see if it changes anything.

radek8, 2020-07-14:
> I'll try to reinstall the firmware to see if it changes anything.

try this
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4920278/BIGTREE_TFT35_V3.0.26.x.zip)

Thijsvanzwam, 2020-07-14:
did that, reinstalled it with the precompiled version and it works now.
strange as I used exactly the same version I installed yesterday... But hey... if it works!

Only thing still is that I have to push the reset button te get onboard SD to work. (the one on the mainboard)
Could that have something to do with the TFT booting faster then the SKR mainboard? (just guessing)

radek8, 2020-07-14:
Boot speed has no effect.
If the display boots, it repeatedly sends a query to Marlin. When Marlin is ready, he will respond with his parameters.

Thijsvanzwam, 2020-07-14:
[Marlin config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4920588/Marlin.config.zip)
Ok, i see. then there could be something wrong in Marlin that doesn't send the correct parameters.
I will review my config files again, put them in as attachment for those who would like to take a look at them.

radek8, 2020-07-14:
If you have Autolevel, get it
#define BABYSTEPPING

radek8, 2020-07-14:
activate
#define AUTO_REPORT_SD_STATUS

Thijsvanzwam, 2020-07-14:
> If you have Autolevel, get it
> #define BABYSTEPPING

I do have autolevel, should I activate this option?

Thijsvanzwam, 2020-07-14:
> activate
> #define AUTO_REPORT_SD_STATUS

ah, forgot that one I see...

radek8, 2020-07-14:
> > Pokud máte Autolevel, získejte to 
> > #define BABYSTEPPING
> 
> Mám autolevel, měl bych tuto možnost aktivovat?

can tune Z-offset when printing

radek8, 2020-07-24:
Please close the ticket

closed 2020-07-24 by Thijsvanzwam