bigtreetech/BIGTREETECH-TouchScreenFirmware #838: A way to control FAN_1 from SRK - Mini - E3 - V2 trough Touch Interface of TFT 35 - E3 - V3

albei, 2020-07-12:
Please add a way to control the fan_1 trough touch menu. The FAN_1 produce a hiss noise when reduce to 50%  ( i got a 12v noctua installed, must keep it at 12v so 50%). And sometimes I just want to turn it off. The only way thats possible is to go in the emulated default menu and change the config settings.

Tygrys-1, 2020-07-12:
In the config.ini set 
`fan_count:2`
Also check the 
`fan_max:....`
Parameter to set max value allowed tor the fans.

albei, 2020-07-13:
I did it, it doesn't work, I just show the Fan_1 available but the settings does not work, cant start the fan or turn it off.
Sadly I can't find a way to mitigate high pitch sound. Looks like Marlin's option for soft_pwm does not apply to controller fan, and the controller fan function has limited options.

GhostlyCrowd, 2020-07-13:
> I did it, it doesn't work, I just show the Fan_1 available but the settings does not work, cant start the fan or turn it off.
> Sadly I can't find a way to mitigate high pitch sound. Looks like [...]

Controller fan is controlled by M710, so unless you tell the touch screen to use that command in the proper syntax it will not work. You will also have to enable #define CONTROLLER_FAN_EDITABLE in marlin source

albei, 2020-07-13:
it is enabled #define CONTROLLER_FAN_EDITABLE. About the M710, I am sorry but I don't understand exactly what you mean.

GhostlyCrowd, 2020-07-13:
M710 is the command used to manipulate the controller fan.  The TFT displays firmware uses the M106 and M107 commands to control fans. those commands are only valid for the Part cooling fan (fan0 typically) now as far as i know.

However Is the Fan1 plug always on? As in you power on the printer and it is on immediatly?

albei, 2020-07-13:
Nope, I mange to control it from marlin interface to work on auto and 50% speed, but this is ads extra steps when i want to turn it off.

GhostlyCrowd, 2020-07-13:
Youre probably going to have to edit the tft firmware to allow it to control your second fan then if you have it working on marlins side.

Tygrys-1, 2020-07-13:
As an alternative you may use custom gcode to send any code needed for Marlin to set the fan. You may use any empty slot in configuration for this and implement it through config.ini. Take a look at [M710 manual](https://marlinfw.org/docs/gcode/M710.html)

Ie. you may set in config.ini:

```
custom_label_8:Fan auto
custom_gcode_8:M710 A1 S255\n
custom_label_9:Fan 50%
custom_gcode_9:M710 A0 S128 I128\n

```

radek8, 2020-07-14:
I understood it correctly?
You have a 24V printer and you are using a 12V fan.
That's why you want him to drive at 50%. And do you mind the sound it makes?

albei, 2020-07-14:
> I understood it correctly?
> 
> You have a 24V printer and you are using a 12V fan.
> 
> That's why you want him to work at 50%. And do you mind the sound it makes?

Yep

radek8, 2020-07-14:
PWM control will always be a bit noisy on the fan.
The best solution is to buy a 24V fan.
You can connect a small value electrolytic capacitor in parallel to a 12V fan. 1 to 10uF, but it will never be completely silent.

radek8, 2020-07-14:
PWM control can also destroy a fan with sensitive electronics

radek8, 2020-07-14:
If you want a quiet fan with PWM use something like this.

https://www.banggood.com/cs/PWM-To-Voltage-Conversion-Module-0-100-PWM-To-0-10V-Voltage-p-1223296.html?gmcCountry=CZ&currency=CZK&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=lijing&utm_campaign=ssc-cz-cs-all-newcustom-ncv80-1118&gclid=EAIaIQobChMI1_qykb3M6gIV2-vtCh3nWQNJEAQYBSABEgISpfD_BwE&cur_warehouse=CN

albei, 2020-07-14:
I'lol go with a 24v silent fan in the end. There are couple of alternatives to noctua, I tested one RS PRO and it's pretty quite. It's a shame that soft_pwm does not apply to fan controller too. It works flawless for FAN_O (parts fan).

radek8, 2020-07-14:
Maybe soft-PWM could be used, but it's about configuring Marlin not TFT display

radek8, 2020-07-14:
But fundamentally, it is wrong to use a 12V fan on a 24V printer

Tygrys-1, 2020-07-14:
Forgive my ignorance - what is the soft_pwm and why it is better? IMHO it is just pulsing 0/24V with different duty cycle?

radek8, 2020-07-14:
https://github.com/Palatis/arduino-softpwm

AVR microcontrollers provide hardware PWM on some pins but if you need PWM on other pins then it must be implemented in software.

It is therefore a SW solution of HW restrictions. The processor has only a limited number of pins that HW-PWM supports. With SW-PWM you can select other switching parameters

albei, 2020-07-14:
It also smother on noise they use low freqvency, and you have extra options. Soft_pwm_scale to change the frequency and sof_pwm_dither to mitigate resolution in case you scale.

Tygrys-1, 2020-07-14:
Ok, thanks. The frequency might be important, I do not know the limits of STM32 hardware PWM. Maybe it will be enough to increase the timer divider on the PWM pin.

radek8, 2020-07-14:
24V fan is the solution here :-)

radek8, 2020-07-23:
Please close the ticket.

closed 2020-07-23 by albei