bigtreetech/BIGTREETECH-TouchScreenFirmware #841: TFT70 Encoder Knob works in opposite direction in 12864 mode.[BUG] (short description)

Madbyte3d, 2020-07-13:
### Description

In 12864 mode the TFT70 encoder knob/button works in opposite direction.  To move down on the menu the knob must be turned counterclockwise. To move up the know must be turned clockwise.  To increase the value of a field (bed temperature for example) the know must be turned counterclockwise and clockwise to decrease the value.

### Steps to reproduce

1. Enter Marlin 12864 emulator mode
2. Push knob to enter menu
3. rotate knob

**Expected behavior**
The opposite.  Clockwise rotation increase a value or moves down the menu.

**Actual behavior**
To move down on the menu the knob must be turned counterclockwise. To move up the know must be turned clockwise.  To increase the value of a field (bed temperature for example) the know must be turned counterclockwise and clockwise to decrease the value.

### Hardware Variant
TFT70 v3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT70_70_V3.0.26.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

AnHardt, 2020-07-13:
To be configured in Marlins Configuration.h
```cpp
/**
 * Encoder Direction Options
 *
 * Test your encoder's behavior first with both options disabled.
 *
 *  Reversed Value Edit and Menu Nav? Enable REVERSE_ENCODER_DIRECTION.
 *  Reversed Menu Navigation only?    Enable REVERSE_MENU_DIRECTION.
 *  Reversed Value Editing only?      Enable BOTH options.
 */

//
// This option reverses the encoder direction everywhere.
//
//  Set this option if CLOCKWISE causes values to DECREASE
//
//#define REVERSE_ENCODER_DIRECTION
```

Madbyte3d, 2020-07-14:
That worked.

Thanks

closed 2020-07-14 by Madbyte3d