bigtreetech/BIGTREETECH-TouchScreenFirmware #842: [BUG] (short description)

Thijsvanzwam, 2020-07-14:
### Description

onboard SD and autolevel not available until reset

### Steps to reproduce

1. switch on power supply, printer starts up
2. wait for "no printer attached" disasters and "ready" shows up
3. All functions are available except onboard SD and autolevel

**Expected behavior**
All function available

**Actual behavior**
When pressing reset in the TFT functions will become available

### Hardware Variant
BTT TFT 3,5 v3.0
### TFT Firmware Version & Main Board Firmware details
Most recent

### Additional Information
Mainboard is BTT SKR 1,4 running on Marlin 2.0.5.3
Issue occurs on marlin bug fix too

Tygrys-1, 2020-07-14:
You may try to turn off this autodetect "feature" ;-)
Put in the config.ini

`onboard_sd_support:1`

See #775 and #770

Thijsvanzwam, 2020-07-14:
did that but no effect

radek8, 2020-07-14:
Do you have this option in Mrlin?
#define SDCARD_CONNECTION ONBOARD

Thijsvanzwam, 2020-07-14:
> Do you have this option in Mrlin?
> #define SDCARD_CONNECTION ONBOARD

yes, it is enabled

oldman4U, 2020-07-18:
Could you remove the display from the metal enclosure and try it again.

Thijsvanzwam, 2020-07-19:
metal enclosure?

oldman4U, 2020-07-19:
How have you mounted the display? Plate, frame, enclosure and is this made
of metal or wood or??

Thijsvanzwam, 2020-07-19:
The display is mounted, but in a printed enclosure made of PLA.

Thijsvanzwam, 2020-07-20:
took it out of the enclosure today but no difference. When I just normally boot up the printer by turning on the power switch, no onboard and autolevel is available. Pressing the reset button and it all works.

oldman4U, 2020-07-20:
I asked because there have been some users with shortage issues installing
this TFT into the Ender 3 standard metal frame causing different  strange
issues.

Thijsvanzwam, 2020-07-20:
I can imagine... I guess it is a software issue. Still trying to find out what the reason is.

oldman4U, 2020-07-21:
Not sure this is software. Have you tried to change the serial cable?

Thijsvanzwam, 2020-07-22:
yep, tried that, two times... one time with original cable and second using Dupont cables. no effect
found out that when I switch from touch mode to marlin mode and back both functions are available. 

When I press reset, is it the same as turning off and on the power switch or is there a difference?
Reset the thing by pressing the button and it works, switch off power and turn it back on and it does not work.

oldman4U, 2020-07-22:
Unfortunately i do not know if this is the same. I am a user like you.

Finally this sounds like a hardware problem to me.

🥺

radek8, 2020-07-22:
Try it with cables to the EXP connector disconnected.

Thijsvanzwam, 2020-07-22:
no luck. tried it but no difference.

traffic-light, 2020-08-01:
I don't know if you use the wifi module. But removing the wifi module solved my problem....

Thijsvanzwam, 2020-08-02:
Yes, I use the Wifi module.
It sounds plausible about the module staying on when doing a reset. So perhaps both the wifi module and the mainboard are simultaneously telling the tft their info, resulting in one of them not being heard at initial startup.
If what you say is the problem, would delaying the startup of the wifi module then be the trick? is there a way of doing that, my knowledge of writing code is quite limited...

radek8, 2020-08-02:
I use a wifi module and I don't have this problem

traffic-light, 2020-08-02:
> Yes, I use the Wifi module.
> It sounds plausible about the module staying on when doing a reset. So perhaps both the wifi module and the mainboard are simultaneously telling the tft their info, [...]

I don't know how the wifi module works. I can take a look at it. I don't know when I have time for it.

For now I don't use it. (The wifi module doesn't give me enough options and upload is slow)

radek8, 2020-08-02:
I also don't use it to print or transfer files, but to test it.
If I need to connect a PC via the Pronterface app.
The SD card is the most reliable for printing

oldman4U, 2020-08-04:
@Thijsvanzwam 

Any news related to the reported issue?

I also use the WIFI module on the TFT also only for Pronterface and play aroud with an SD Cloud at the moment.

Thijsvanzwam, 2020-08-04:
Nope, no news yet. 
I really think that some way of delaying the startup of the wifi module would make a difference. Just cannot find a way of doing that. (because my lack of knowledge about writing this kind of code)
For now I just accept it.
Autolevel is done by the startcode of each print so that just works. 
I use the SD card in the TFT, the one on the mainboard is only used for flashing the firmware now and the wifi I hardly use because of its limitations. 

Hopefully the wifi gets more options in the future and perhaps an option to delay startup as well...

traffic-light, 2020-08-04:
I found a command that you can use when you connect to your esp via a terminal.

````
* Delay command
[ESP290]<delay in ms>[pwd=<user password>]
````

I'll try this. When this works I will give a short guide how to do it.
I hope it works. Sometimes it's ease, for example when using the m503 command, to use the webui.

traffic-light, 2020-08-04:
What version do you use on your ESP?
I use 2.1.0.c26.

When u use atleast version 3.0 you can change it in the settings of your esp3d

Link to solution: https://github.com/luc-github/ESP3D/issues/329

oldman4U, 2020-08-04:
I am also using v2 and failed to upload v3 because of me i guess;-) But i am able to use wifi and abl without reset so not sure why i would need a delay to get it work.

traffic-light, 2020-08-04:
Because we have that issue :stuck_out_tongue_winking_eye:

oldman4U, 2020-08-04:
Which one😎

But i believe you are looking for the wrong solution. Which serial ports have you defined and in which order? I use only one and this is „2“.

traffic-light, 2020-08-04:
I've placed the WiFi module in de dedicated port on my TFT35-E3-V3.0

And serial ports on the mainboard or on the tftscreen?

oldman4U, 2020-08-04:
Me2

Mainboard

um 16:30:

traffic-light, 2020-08-04:
Okay. These are my Serial settings:
````
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0
````

These where recomended

radek8, 2020-08-04:
@Thijsvanzwam 
Can you disconnect WIFI and see if the problem persists?
Something that is probably not related to your problem is being addressed here

Thijsvanzwam, 2020-08-04:
I'll try that later, not at home right now. enjoying the good weather on the boat at the moment...

traffic-light, 2020-08-04:
@radek8 I don't know if @Thijsvanzwam is getting the same. But when I switch to **Accespoint** mode instead of **Clientstation** I don't need to reset for ABL etc.

oldman4U, 2020-08-04:
On my SKR E3 DIP I use the following serials.

#define SERIAL_PORT 2
 */
//#define SERIAL_PORT_2 -1

I can print from On Board and TFT SD card and USB stick, have the WIFI connection and Autolevel.

It is a long time ago and I can not remember why I finally used this setting, but it works. Maybe another setting is needed for the mini, but at least it is worth to give it a try.

Thijsvanzwam, 2020-08-26:
nope, tried all but no luck.
Just accepted it for now...

closed 2020-10-26 by Thijsvanzwam