bigtreetech/BIGTREETECH-TouchScreenFirmware #843: [BUG] Fan Speed is not updated during print

Pattern91, 2020-07-14:
### Description

When the Fan Speed of the part cooling fan is changed by Gcode it ist not updated on the status screen nor in the fan setting menu. 

### Steps to reproduce

1. Start print from USB-Stick
2. wait for Fan Speed adjustment (after first layers)

**Expected behavior**

The fan speed in the Status Screen should be the same as the actual fan speed of the part cooling fan. 
**Actual behavior**

The fan speed is always at 0%
### Hardware Variant

TFT35
### TFT Firmware Version & Main Board Firmware details

TFT Firmware: Vx.x.26
Marlin: 2.0.5

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2020-07-15:
Can you compile a new FW (do not use precompiled)?
Or use this one:
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4923346/BIGTREE_TFT35_V3.0.26.x.zip)

Pattern91, 2020-07-15:
I compiled it on Friday (10th July) but I can do it again.

radek8, 2020-07-15:
Several fixes have been uploaded since Friday

Pattern91, 2020-07-15:
I updated the firmware to the version, you send me. Here is an image of the status screen during the print. the cooling fan is spinning. If I manually control the fan via the tft it shows the correct percentage.
[image]

radek8, 2020-07-15:
For me, the described error does not manifest itself.
Can you upload a print file that has a problem?
Thank you

Pattern91, 2020-07-15:
[tft35 front.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4926370/tft35.front.gcode.zip)
 For example this one. It is sliced via Cura 4.6.1

radek8, 2020-07-15:
In the second layer, the fan speed was automatically set for me, and displayed on the screen.
For a change, I found that my print speed was not updated
It's mysterious ....

radek8, 2020-07-15:
I found a problem updating the value with print speed.
Use this FW
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4928048/BIGTREE_TFT35_V3.0.26.x.zip)

radek8, 2020-07-15:
If the previous FW doesn't help, try this one. It is with another adjustment.
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4928321/BIGTREE_TFT35_V3.0.26.x.zip)

jnle7, 2020-07-16:
I have the exact same issue ... part cooling fan is not updated on the LCD screen after the 2nd layer ... fan runs at the correct speed but speed is always 0 in LCD display ...

radek8, 2020-07-16:
> Mám přesně stejný problém ... rychlost chlazení ventilátoru se na LCD obrazovce po 2. vrstvě neaktualizuje ... ventilátor běží na správné rychlosti, ale na LCD displeji je vždy rychlost ... Mám TFT35 E3 v3 ...

You can try both FWs for the test.
They are the same as the TFT 35 E3 V3, only the encoder wheel will not light up.

radek8, 2020-07-16:
If you (jnle7 and Pattern91) have compiled your own FW, or used mine, it's weird that you still have the same problem.
For me, your problem does not manifest itself.
Can you send the config.ini you used to configure?
I will compare how we have a different configuration.

Pattern91, 2020-07-16:
Thanks for your firmware but it didn't work... even if I change the fan speed via the "Marlin" Interface (the not touch Interface), the fan speed is not updated in the Touch Mode. Attached you find my current config
[config.ini.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4932639/config.ini.zip)

radek8, 2020-07-16:
> 
> Edit: during the print. The second firmware worked 😋

Super
I found a problem
I have an option in config.ini
"Serial_always_on: 1" - that's why my fan speed has been updated.
You have "Serial_always_on: 0". that's why 1 FW didn't work for you, but the other one was fine.
I will suggest a fix to make the fan speed update work for both options.

PR #847 (2020-07-18): Fixed automatic update of fan speed
Fixed automatic update of fan speed to status screen when printing if the "Serial_always_on: 0" option is activated

#827 was a bad fix which produced problems with updating values on the status screen
replace it patch #840, #846, #847
 
resolves #843
closed 2020-07-18 by bigtreetech