bigtreetech/BIGTREETECH-TouchScreenFirmware #845: Hardcoded M18 in printing.c makes the end gcode partial useless

oldman4U, 2020-07-15:
It is not possible to keep a certain motor active after the print, because a M18 in end gcode will be overwritten by this in printing.c

 // disable all stepper
  mustStoreCmd("M18\n");

I believe this part of printing.c should be ignored, if the user defines an m18 in the end gcode. Maybe this should also be the case with the heater and the fan?

Thank you

Tygrys-1, 2020-07-20:
This is more complicated issue. It should be aware of the M84, M18, know the EndGCode and CancelGCode and know the reason of the printer going idle to examine the right GCodes. It will require to rethink and rewrite a printing logic. It will not solve the problem if the M18 or M84 were invoked from the gcode file, as i.e. my Cura is sending M84 X Y E; at the end of the file, for holding the Z.

It might be simpler to put in the config.ini something like "Default motor release" and "Default cooldown" gcodes, and invoke them instead of hardcoded ones. Or just ignore hardcoded values if the custom EndGCode is enabled.

oldman4U, 2020-07-20:
Hello Tiger.

Whatever the solution will be finally. It does not make sense to offer a
function which is finally overruled especially in a case like this where
the overruling part is not „available“ to the average user (like me)

Tygrys-1, 2020-07-20:
Maybe the software was first build with hardcoded go-idle function, then the start/end gcodes were added for flexibility and now the old function should be rebuilded with consideration of this addition. 

You always may delete this M18 line and compile your own version of firmware ;-)
I do not see the reason to auto-release steppers after the print job - they release themselves either way after some time, and if someone wants to release them by hand it always can be done with menus.

oldman4U, 2020-07-20:
I could not agree more. Could you please change it and make a PR. This
would be great.

Thank you

Tygrys-1, 2020-07-20:
It is not that simple.
I think it requires more discussion. Someone put M18 there and if I do not know the reason, I cannot delete it, as it might have consequences I'm not aware of. Also, maybe it should be not a hard delete, but conditional if no endcode is enabled.

RetiredTexan, 2020-07-21:
I concur with oldman4U,  M18 should never be issued in hardcoded logic.  All decisions of the type should be in configure files.  In my case,  my cnc router freefalls to the table when the z motor is released.  My router is turned on and off manually.  With the current code, my router will drill a hole through the table at the end of the job, not cool.  I have to resort to putting pauses in the gcode to prevent this.  I spent a week chaind down what was causing this behavior and found it in printing.c.  I have commented out the offending code, but at the expense of a weeks time searching for an undocumented hidden bug(that's what it is in my opinion).  And had to recompile code to top it off.

jeffeb3, 2020-07-22:
I've been poking at this code, and I have to say that the M18 seems really out of place. 

We should be striving for the "least suprise" and having it disable the steppers ever, when it isn't in one of the ending or other gcode sequences in the config.ini is really surprising to me.

I would really rather all of the gcode when going to idle be one configuration value, and the default should include hotend to 0, bed temp to 0, and nothing else. M18 is really awful on machines where stuff needs to actively be held up. If your machine doesn't have that problem, fine, but does it do any harm to keep the motors energized until Marlin times out, when going to idle?

I'd be happy to PR a change to make that entire idle function come straight from the config.ini file. If you think it will get merged I can do that.

oldman4U, 2020-07-22:
Hi Jeff

It would be great if you could do the PR. Do you think we should change the
current settings in config.ini as well?

bigtreetech, 2020-07-23:
Hello, will move all the end gcode to config.ini file later

oldman4U, 2020-07-23:
Thank you

Should I close this ticket now or will you do so automatically with the fix?

bigtreetech, 2020-07-23:
I will close here when fix is merged

bigtreetech, 2020-07-28:
deleted the built-in end gcode in #875, so close here

closed 2020-07-28 by bigtreetech

oldman4U, 2020-07-28:
👍🏻