bigtreetech/BIGTREETECH-TouchScreenFirmware #848: TFT35 E3 V3  not see onboard SDcard

nesergen, 2020-07-17:
### Description
On May 31, I made an order for a TFT35 E3 V3 display and an SKR 1.4 turbo board. After receiving, I connected them to the printer. The display and board connected to each other without issue. The flash drives on the board and on the display have been identified. However, there is a problem with the motherboard SD card. The display sees it, opens it, but after opening it freezes, stops responding to touches. Only one touchpoint works, when you click on it, an empty window with the title ERROR appears for a moment. Only a long touch helps, which takes the display to the operating mode selection menu. 
I assumed that this is a defect in the old factory firmware for the display. I downloaded the new firmware, installed it, the problem remained. The problem occurs when the display parameter "onboard_sd_support: 2" is set. When changing the parameter to auto-detect SD card ("onboard_sd_support: 1"), The SD card not detected at all.
After reading the forums, I updated the Marlin firmware with the latest bugfixes. But the problem remained.

### Steps to reproduce

1. Marlin Bugfix CONFIGURATION_H_VERSION 020006
2. BIGTREE_TFT35_V3.0_E3.26.x firmware
3. config.ini --> "onboard_sd_support: 1"
4. use the display to select the flashcard on the motherboard.

**Expected behavior**
Standard navigation through the list of files on the sd card

**Actual behavior**
The display freezes.

### Additional Information
[https://cloud.mail.ru/public/3B1S/4skRYcEQv](url)
[https://cloud.mail.ru/public/7i8F/2TqKCHxRM](url)
[https://cloud.mail.ru/public/2js3/4HUSumbwg](url)

Tygrys-1, 2020-07-18:
Can you use the SD card in Marlin mode?

nesergen, 2020-07-18:
> Can you use the SD card in Marlin mode?

Yes, I can do it.

nesergen, 2020-07-18:
Small clarification. Once again I downloaded Marlin's firmware with corrections, added my settings for the printer to it. Now the SD card is visible on the display during auto-detection, but the error still appears when opening the card.
What the error looks like.
1. I click on the motherboard SD card icon
2. "Loading ..." appears.
3. An empty error window appears
4. The SD card directory tree appears. But I cannot navigate through it.
The only work point is the area where the "OK" button is located on the error window. It seems that the window with the error has disappeared, but it still takes all touches on itself. When you click on the area of ​​the "OK" button, the window appears for a moment and disappears again, but does not allow touches to reach the list of files.

radek8, 2020-07-19:
You have to format your SD card.

Tygrys-1, 2020-07-20:
I think that the file with dot-and-something may be read as an empty file name with a long extension and this fouls the directory reading procedure. It might be an exception or other unpredicted behavior. You may try to investigate further with filename like this and find the pattern.

nesergen, 2020-07-23:
The problem resolved itself. No error occurs. There may have been problems in the file system of the card. The problem supposedly went away after deleting the ".Trash-1000" folder. Nevertheless, this folder is again present on the card, but does not cause problems. Folders with a dot at the beginning are detected without problems.

closed 2020-07-23 by nesergen

Tygrys-1, 2020-07-23:
Good to know.