bigtreetech/BIGTREETECH-TouchScreenFirmware #850: [FR] Filament Change/Offload/Load features

ShaggyDog18, 2020-07-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Impossible to offload/load filament from a touch screen. Have to jump to emulation mode and use Marlin menu to perform Filament change tasks.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Deploy Filament change features as they are in Marlin:  
- have to define distance for the filament to go from extruder to the hotend (in mm) and 
- temperature for extrusion for different standard materials (may have same set as in Marlin: PLA, ABS, etc..)
- three menu items:
  - Change filament (performs offload and waits for the load, essentially, a combination of the two below)
  - Offload filament - pre-heat hotend to a pre-defined standard temperature for a selected material and unload a pre-defined length of filament
  - Load filament - pre-heat hotend and load a pre-defined length of filament with an option of slow filament delivery at the end to complete the load by manual control...

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Tygrys-1, 2020-07-20:
TFT can talk with the Marlin only through the GCode, but required feature requires some kind of interaction.
Please check the #769 for workaround..

oldman4U, 2020-09-05:
@ShaggyDog18 

Please close this ticket as it is covered in the pinned ticket from traffic-light.

Thank you

ShaggyDog18, 2020-09-06:
closed as it is covered in the pinned ticket from traffic-light.

closed 2020-09-06 by ShaggyDog18