bigtreetech/BIGTREETECH-TouchScreenFirmware #852: If I set #define SERIAL_PORT 0 I get this error message any Ideas what is wrong?  The terminal process "C:\Users\gegro\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BIGTREE_GTR_V1_0'" terminated with exit code: 1. It will work if I set #define SERIAL_PORT -1 The other #define SERIAL_PORT 2-0 seams to have no effect[Q] (Your Question?)

Gordon44, 2020-07-20:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

oldman4U, 2020-07-21:
Hi.

This seems to be GTR1 related, so I would recommend to use the GTR repository. This is for BTT Touchscreen only. Please do not forget to close this ticket here

Thank you

bigtreetech, 2020-07-22:
Hello,  the GTR serial port is start from 1, so 0 is an incorrect setting， Please refer to GTR's user manual to find and set up the correct serial port you want to use. 
It's not about TFT, so it will be closed here, but if your problem is not solved, you can contact me here

closed 2020-07-22 by bigtreetech

oldman4U, 2020-07-22:
Hi BTT.

Is there a way to close the other unused tickets also? I marked some of
them but „you“ did not close it in this repo, the tft35 and tft35 dip, the
skr pro as well as the skr e3 dip repo.

I have also another development question. Is there a mail address i could
use?

Thank you