bigtreetech/BIGTREETECH-TouchScreenFirmware #855: Information: Installation of the Cura Preview plugin on macOS.

oldman4U, 2020-07-21:
Installation of the Cura Plug-In on macOS works a little bit different.

On Finder level navigate to Cura (usually in Applications).
Make a right mouse click on Cura and select "Show Package Contents".
This opens the package and shows a folder named "Contents" - double click on it.
This opens the Contents folder and shows several folders - double click on the folder named "plugins" to open it.
This opens the plugins folder and shows another folder named "plugins" - double click to open it.
Copy the Bigtree3DPlugin folder which you downloaded before into that folder.

(Re)start Cura.

This is really a great feature, I like it!

blackus3r, 2020-07-21:
Where can I download this plugin?

oldman4U, 2020-07-21:
It is the whole folder not a single file.

Tygrys-1, 2020-07-22:
Download plugin [here](https://github.com/bigtreetech/Bigtree3DPlugin)

ssombra, 2020-07-28:
The documentation of the plugin says it works starting on Vx.x.27, but the last release is x.x.26, am I wrong?

oldman4U, 2020-07-28:
First RELEASE version will be 27. current v26 has it also.

oldman4U, 2020-07-28:
Happy printing

closed 2020-08-15 by oldman4U