bigtreetech/BIGTREETECH-TouchScreenFirmware #858: LCD Panel did not work when add the wifi adapter to SKR 1.4

Talha909, 2020-07-22:
I have SKR 1.4 Tubo with TMC 2130. I also attached the TFT 24.  I have installed the Marlin version.  Marlin mode on TFT screen working good. But in touch mode there is error message Printer not attached. 
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4958371/Marlin.zip)
[image]

Tygrys-1, 2020-07-22:
Marlin has 2 serial ports. One is for USB, the other is for TFT **or** WiFi. You may have the fish or the aquarium.

Talha909, 2020-07-22:
but fish lives in the aquarium.

Tygrys-1, 2020-07-22:
> 
> 
> but fish lives in the aquarium.

In generall they live in the water :-). That is kinda joke: what the father said to daughter asking for goldfish ;-)

In this case you can not have them both, can have either one. One port, two functions - that does not work. 
It is not the TFT bug, but the Marlin's "feature".

However, here is a way to have them both:
If you can sacriffice the SKR's USB port, you may configure the SERIAL_PORT to the real serial port number of TFT and SERIAL_PORT_2 to the real serial port of WiFi. -1 means "autodetect port with the device", 3 is USB on SKR.
You will got your goldfish ;-)

radek8, 2020-07-22:
Replace the USP port with a virtual COM port over Wi-Fi.
See #704

oldman4U, 2020-07-24:
Hello @Talha909 

Do you know what you have to do and that this is a limitation of available ports and not a bug? Please close this ticket. Thank you

bigtreetech, 2020-07-28:
Hello, As mentioned above, Marlin firmware currently only supports two Serial ports, while USB, WiFi and Touch screen all need to use Serial ports, so at present, you can only choose two of this to work.
And it's not a bug,  so close here,  If you have any other questions, please contact me here

closed 2020-07-28 by bigtreetech