bigtreetech/BIGTREETECH-TouchScreenFirmware #859: Resume after filament runout.

Thijsvanzwam, 2020-07-22:
I connected the filament runout sensor (switch-type) to the TFT3,5. When filament runs out it works flawless. Nozzle moves up and out of the way, resume button appears on the screen.

When I press resume the nozzle moves back to the print, poops some filament and continues the print.

But the it moves up again, saying the filament is runout and moves out of the way again.

When I press "pause" during a print and the press resume it works just fine...

added my config files.

Can anyone have a look what I might be missing?

[Archief.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4960283/Archief.zip)

bigtreetech, 2020-07-23:
Hello, This is a bug and it should be fixed in #857

gordo3di, 2020-07-24:
Confirmed, works great now.

Thijsvanzwam, 2020-07-24:
you mean it works in the latest firmware?

gordo3di, 2020-07-24:
Yeah they fixed it with this commit. Basically after it triggers it sets the runout back to false. That was missing.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/857/commits/22dfa8566d371e26d8e24c297f0da8d6e6b2c490

Thijsvanzwam, 2020-07-25:
I agree.
tested with latest firmware and it is fixed.
I'll close the ticket.

Thijsvanzwam, 2020-07-25:
only thing now is the big blob of filament deposited on the point it resumes...
a filament purge in a corner would be nice...

gordo3di, 2020-07-25:
@Thijsvanzwam Agreed. I'll look into fixing that today. It's very annoying.

Thijsvanzwam, 2020-07-25:
that would be great!
I guess that a small introline somewhere in the right corner (when facing the printer) would be nice, as the initial introline in most start code is done on the left.

gordo3di, 2020-07-25:
@Thijsvanzwam @bigtreetech 

Lol that was easy. I would submit a commit however i'm still trying to figure out github and using it with VS code.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/866

drlohwk, 2020-08-17:
Which firmware version should I use? Mine still cannot resume printing and keep triggering filament runout. Also a big blob of plastic deposited on the print. Can u please point me to the right direction? Thanks.

Thijsvanzwam, 2020-08-17:
I use the latest self compiled.

oldman4U, 2020-10-09:
Hi. What is the status of the ticket?

closed 2020-12-15 by stale[bot]