bigtreetech/BIGTREETECH-TouchScreenFirmware #866: [FIX] Print resume purges filament on model

gordo3di, 2020-07-25:
### Description

Filament Should purge in the parking position after resuming a print. Instead it purges on the print itself.

### Steps to reproduce

1. Pause print
2. Resume print

### Hardware Variant
TFT35_E3_V3 and TFT35_V3

### TFT Firmware Version & Main Board Firmware details
20200721

### Solution

In Printing.c you need to move some code around so the purge happens before the extruder moves back to the model. Also added an M0 to wait for user after purge completes. We tried creating a dialog box to pop up on the screen however it wont purge and pause since the queue hasn't restarted yet. The M0 seems to do the trick

Line 216

```
//gcreate move code here
        if(heatGetCurrentTemp(heatGetCurrentToolNozzle()) > infoSettings.min_ext_temp)
        {
          mustStoreCmd("G1 E%.5f F%d\nM0\n", tmp.axis[E_AXIS] - infoSettings.pause_retract_len + infoSettings.resume_purge_len, infoSettings.pause_feedrate[E_AXIS]);
        }
//
        if (coordinateIsKnown())
        {
          mustStoreCmd("G1 X%.3f Y%.3f F%d\n", tmp.axis[X_AXIS], tmp.axis[Y_AXIS], infoSettings.pause_feedrate[X_AXIS]);
          mustStoreCmd("G1 Z%.3f F%d\n", tmp.axis[Z_AXIS], infoSettings.pause_feedrate[Z_AXIS]);
        }
//gcreate code was here
```

Zowen79, 2020-07-27:
Not sure if this is the area for this, but I think I found an issue in Settings.c.   

114: infoSettings.pause_pos[X_AXIS]      = NOZZLE_PAUSE_X_POSITION;  // X
115: infoSettings.pause_pos[Y_AXIS]      = NOZZLE_PAUSE_X_POSITION;  // Y

I am pretty sure that NOZZLE_PAUSE_X_POSITION on Line 115 should be NOZZLE_PAUSE_Y_POSITION. I don't know what impact this would have, but I would guess that the infoSettings.pause_pos[Y_AXIS] will be set to the same location as the X_AXIS. In my case I set my pause position to X185 Y480 so I can get to the extruders easier. Hope this helps

Zowen79, 2020-07-27:
Also in Leveling.c
79: void moveToLevelingPoint(u8 point)
80: {
81:  s16 pointPosition[4][2] = {
82: {infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge, infoSettings.machine_size_min[Y_AXIS] + infoSettings.level_edge},
83: {infoSettings.machine_size_max[X_AXIS] - infoSettings.level_edge, infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge},
84: {infoSettings.machine_size_max[X_AXIS] - infoSettings.level_edge, infoSettings.machine_size_max[Y_AXIS] - infoSettings.level_edge},
85: {infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge, infoSettings.machine_size_max[Y_AXIS] - infoSettings.level_edge},
86:  };
Line 83 doesn't follow the same pattern as the 1 above it, and the 2 below. This one I am not sure if it will have an impact. Just putting it out there, if it is a problem, great find, if not, then no harm no foul. Thanks

gordo3di, 2020-07-27:
@Zowen79 Nice catch. How did you track them down?

Interesting. It looks like this is what sets the pause location from the configuration.c variables so it should not work right. However, the pause/runout location seems to work fine from the positions I've set in my config file. It goes to X0 Y210 I wonder that this affects then?

Also ditto on the Leveling.c I think my leveling has been a bit off and this could explain why... Gonna try recompiling right now

Zowen79, 2020-07-27:
When I was pausing it wasn't going to the location I set, so I did a search in VS for axis and went thru all of the results. Tedious, but worth it. Glad I could help

Zowen79, 2020-07-27:
I made the changes to my local build. So we shall see if it makes a difference

closed 2020-08-24 by gordo3di