bigtreetech/BIGTREETECH-TouchScreenFirmware #867: [BUG] (Parameters menu does not allow to set correctly some parameters + Numpad.c can cause a crash)

digant73, 2020-07-25:
### Description

The Parameters menu does not allow to correctly set all the parameters such as "units-per-steps".
E.g. the "units-per-steps" parameter value is configured in Marlin as "445.00".
The Parameter menu display the correct value but does not allow to change the value providing 2 digits after the "." character.
E.g. I can only set the value "436.2" instead of the desired value "436.27).

### Steps to reproduce

1. [First Step]
Open Parameters menu
2. [Second Step]
select units-per-steps
3. [and so on...]
try to set the value 436.27

**Expected behavior**
value 436.27 is set

**Actual behavior**
only value 436.2 can be set. Second digit (7) after the "." character cannot be set

### Hardware Variant
pin_MKS_TFT32_V1_4
### TFT Firmware Version & Main Board Firmware details
TFT 0.26.1 (current main branch), Marlin 2.0.5.4

### Additional Information
I found the problem is in the Numpad.c file. The function numPadFloat allow to set 5 digits while the value I need to set requires 6 digits.
I also found that the buffer holding the value can be overflowed causing a crash.
I fixed both bugs:
1) I provided 2 different buffer lenghts for the functions numPadFloat and numPadInt.
The lenght of the buffer used for int value remained the same (6) while the lenght for the buffer for float values was set to 7 (allowing a maximum of 6 digits)
2) Furthermore I fixed ParameterBuf and tempstr buffers instanciation. Their size were not correctly set and could cause a crash

Attached the zip file with the fixes.

@bigtreetech please have a look to the code and possibly merge these fixes in your next release

[Numpad_fix.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4975839/Numpad_fix.zip)

radek8, 2020-07-25:
If you know how to fix a bug, create your own PR
It will speed up development and help everyone who can't do it like you. 
You're good and it's a shame not to use all your ideas :-)

digant73, 2020-07-26:
I will create a PR with my additions. I only provide icons for TFT28 Unified Menu. So I designer providing icons for all the other TFTxx and Classic Menu will be needed

radek8, 2020-07-27:
I tested your Numpad patch and it really works.
Will you create a PR?
I would publish PR, but I don't want to take your job or interfere with your unfinished PR

digant73, 2020-07-28:
> I tested your Numpad patch and it really works.
> Will you create a PR?
> I would publish PR, but I don't want to take your job or interfere with your unfinished PR

Not at the moment. I'm improving features such as:
* ProbeOffset
* BabyStep (the save button can be temporary enabled just to be used in the Leveling menu and not when printing)
* Adding a notification window that is used just to display for few seconds a notification not requiring any user interaction
* Adding in config.ini a notification type parameter that will set the type of notification window (none, notification and dialog) to be used to display all the echo messages not provided in the echo list (in ParseAck.c)
* Assigning the notification type also to the echo list in ParseAck.c so you can differenciate the type of notification window applied to any echo string in the list (none, notification and dialog). That notification type can be changed. E.g.  just to temporary suppress an echo message (e.g. in my new ProbeOffset menu)
* PID autotune menu allowing to notify the user when the process is successfully completed or aborted
* cleanup here and there
However I need more time before publishing the PR

You can publish your PR. I will have I look to that and probably I' will provide my changes as I made with guruatwhal.
If you publish your PR, include this bug fix

PR #878 (2020-07-31): Numpad repair
Fixed entering values via numpad if the value contains 5 digits - resolves #867
typo correction  - mentioned #866
Add a Save button to the BabyStep menu  - mentioned #869
closed 2020-07-31 by bigtreetech