bigtreetech/BIGTREETECH-TouchScreenFirmware #869: [FR] Change icon

DrumClock, 2020-07-25:
### I would suggest changing the icons in "Z offset" and "Babystep" modes.

**I would replace the original**  

[image]
[image]

 **with   "NOZZLE"  icons**

[image]
[image]

### for better orientation

bryan065, 2020-07-26:
Icons can probably be changed as part of #863 as well?

DrumClock, 2020-07-26:
**How to change icons?**
I downloaded the current version for TFT 50 V3.0 compiled in VSC and the` .bin `file updated the display.

Tygrys-1, 2020-07-26:
Make a folder on SD card called TFT50. Copy there the content of the "Copy to SD Card root directory to update....."\TFT50 folder, change the icons you have in the zip with corresponding ones in this folder and boot the printer with the SD in the TFT slot.

DrumClock, 2020-07-26:
Hi, 
Okay, but it replaces the icons in all menus where there are "+/-"

**I just want to replace it in the "Babystep" menu.**
Is there no need to change something directly in the FW TFT50?

[image]

Tygrys-1, 2020-07-26:
For this you'll have to modify the firmware. You have to add the icons using unique names to the icon list and to the icon folder, and modify the menus to display your icons. The icon space is limited and I do not know, if we can add new ones or how many of them. This is to be checked.

DrumClock, 2020-07-26:
### I am not a programmer, can you please check it and add it if necessary? 
If it could, could this be added as well?

[image]

### I would replace and test the modified files.
Thank you

Tygrys-1, 2020-07-26:
Sorry to say, I'm not that skilled to do this.

DrumClock, 2020-07-26:
### OK, maybe someone will contact me who can handle it :)

bryan065, 2020-07-26:
Hey @DrumClock, I've added an optional `#define FRIENDLY_PROBE_OFFSET_LANGUAGE` TO #863. You'll be able to use it once it gets merged

DrumClock, 2020-07-27:
Hey @bryan065 
in the `"Z Offset"` menu? I meant in the `"BabyStep"` menu.
**If I now change the value in the "Z Offset" menu, the Hotbed does not move.** 
However, if I do it via the "BabyStep" menu, it can be seen immediately.

DrumClock, 2020-07-27:
I have tried your FW `pr/1` for TFT50 and it looks very nice. 
**At UBL, I would just warm up HotBed to 70 degrees first.**

I use two HotEnd on V-extruders and a Touch MI probe.
https://www.youtube.com/watch?v=BOaIMiyzaks
https://www.youtube.com/watch?v=DsCdpfjLS2g

I quite miss the TFT menu from the LCD 12864.
Could it be added to the configuration?
[image] 
### TOOL OFFSET
[image]
### TOUCHMI
[image]

radek8, 2020-07-27:
I agree to return the sabe offer in the Baby-step offer.
It was originally there and someone removed it

DrumClock, 2020-07-28:
Hi, @radek8 
The **Z Offset** value is saved, but the value does not change in the BabyStep menu.
After pressing the reset the icons will be overwritten incorrectly  ....**. it's confusing ....**

radek8, 2020-07-28:
That is right
The Z Offset value is saved. Baby Step is added or subtracted from Z Offset
Baby Step always starts at zero

DrumClock, 2020-07-28:
@radek8 
### That's not good ... it's confusing.

[image]
[image]
[image]

radek8, 2020-07-28:
That's not good. :-(
I'll try to fine-tune it in the evening

radek8, 2020-07-28:
The fact that the Z Offset is not saved could also be caused by the old version of Marlin.
How old is your version of Marlin?

DrumClock, 2020-07-28:
**Marlin-bugfix-2.0.x  /  Downloaded: 16.7.2020**

DrumClock, 2020-07-28:
Hi @radek8 
Once you've started it, you should finish it.  :)
It looks like Digant73 will release its update right away. :(

radek8, 2020-07-28:
> Hi @radek8
> Once you've started it, you should finish it. :)
> It looks like Digant73 will release its update right away. :(

@DrumClock
I added a Save button to the BabyStep menu.
Can you try it?
#878

DrumClock, 2020-07-29:
> @DrumClock
> I added a Save button to the BabyStep menu.
> Can you try it?
> #878

Umíte česky ?

radek8, 2020-07-29:
> > @radek8
> > I added a Save button to the BabyStep menu.
> > Can you try it?
> > #878
> 
> Umíte česky ?
Ano :-)
Yes, but the local community will not like it if we use a language other than English

DrumClock, 2020-07-29:
> > > @radek8
> > > I added a Save button to the BabyStep menu.
> > > Can you try it?
> > > #878
> > 
> > 
> > Umíte česky ?
> 
> Ano :-)
> Yes, but the local community will not like it if we use a language other than English

I understand that, can we talk Czech together somewhere?

radek8, 2020-07-29:
radek.zobalzavinacptacek.cz

DrumClock, 2020-07-29:
thanks, I have already sent an e-mail .... did not come?

DrumClock, 2020-07-30:
> Hey @DrumClock, I've added an optional `#define FRIENDLY_PROBE_OFFSET_LANGUAGE` to #863. You'll be able to use it once it gets merged
> 
> [image]

### Hi @bryan065 
**branch `pr/1` means flawless.**
I would just add Hotbed heating for ABL.

closed 2020-08-01 by DrumClock